beltransen/velo2cam_calibration #2: undefined reference to `pcl::SampleConsensusModelCircle3D<Velodyne::Point>

anant3110, 2017-05-24:
CMakeFiles/laser_pattern.dir/src/laser_pattern.cpp.o gets an error while compiling.
All the references to pcl::SampleConsensusModelCircle3d and its member functions are getting an undefined reference.

beltransen, 2017-05-24:
Hi, can you please give us more info about the console output? We also need to know how are you compiling the repo, and if you have all the required dependencies installed. Which ROS version are you using?

anant3110, 2017-05-24:
this is the attached log. I am using Indigo.
[log.txt](https://github.com/beltransen/velo2cam_calibration/files/1025476/log.txt)

beltransen, 2017-05-24:
In the log there is no error, neither pcl::SampleConsensusModelCircle3D is never mentioned. Maybe this is not the full console log?

anant3110, 2017-05-24:
Sorry, My bad!
[log1.txt](https://github.com/beltransen/velo2cam_calibration/files/1025508/log1.txt)

cguindel, 2017-05-25:
Could you please check if the PCL variables are correctly set by the CMakeLists when you run catkin_make?, namely PCL_LIBRARY_DIRS, PCL_DEFINITIONS, etc. It seems to be a problem with your PCL setup.

anant3110, 2017-05-25:
I checked out the CMakeLists.txt and everything seems to be fine. The problem here is that every other cpp file is getting compiled except laser_pattern.cpp

anant3110, 2017-05-25:
[CMakeLists.txt](https://github.com/beltransen/velo2cam_calibration/files/1028591/CMakeLists.txt)

beltransen, 2017-05-25:
Could you please show what's inside the PCL variables while compiling? This can be done through by calling the MESSAGE() function in the CMakeLists file. The package is working fine in Ubuntu 16.04 and ROS Kinetic, as validated by ROS repository indexer, so it is likely a compatibility issue with your PCL setup.

anant3110, 2017-05-29:
Is there anyway that you can propose that it works with ROS Indigo and Ubuntu 14.04? 
Our project is working on that version only.

beltransen, 2017-05-29:
We have just compiled the package in an Ubuntu 14.04 computer with ROS Indigo and we got no errors. Make sure you have installed all the dependencies successfully or try to compile it on another computer, as it seems to be all right. Please, keep us up to date with any progress

anant3110, 2017-06-02:
I ended up using Ubuntu 16.04 only.

closed 2017-06-02 by anant3110