beltransen/velo2cam_calibration #3: cannot launch node of type *

anant3110, 2017-06-02:
After Doing
`roslaunch velo2cam_calibration stereo_pattern.launch'
A lot of errors pop up saying that nodes were not found. i.e.
[disparity/disparity_node], opencv_apps/edge_detection], pcl_utils/disp_masker], [calibration_from_plane/plane_segmentation].

cguindel, 2017-06-02:
Sorry, you are totally right; we were using our own unreleased dependencies. Thank you for the heads-up. 9abff1a7f6535df6a896219724fc51310288aa5c should fix it. You will also have to install [opencv_apps](http://wiki.ros.org/opencv_apps) from the ROS repositories:
```shell
sudo apt-get install ros-kinetic-opencv-apps
```
Please, check it and tell us if there is anything else wrong.

anant3110, 2017-06-02:
It works! Thank You so much!

closed 2017-06-02 by anant3110