beltransen/velo2cam_calibration #5: Plane segmentation node in Ros Jade

BrennoCaldato, 2017-12-05:
Hi,

I'm trying to run this package in Ros Jade and the node v2c_plane_segmentation is crashing without any apparent reason, it just prints "Segmentation fault (core dumped)". 

I know that this package has no official support to Ros Jade but I can not upgrade my machine to accept Ros Kinetic. Has anyone had this issue and have a solution?

Best regards

cguindel, 2017-12-13:
Unfortunately, we do not have a ROS Jade machine available to reproduce the error, but the v2c_plane_segmentation node is very straightforward, and it shouldn't require significant changes between different ROS versions. As you may have noted, it is basically a ROS wrapper for [this example](http://pointclouds.org/documentation/tutorials/planar_segmentation.php), which is already implemented by [pcl_ros](http://wiki.ros.org/pcl_ros/Tutorials/SACSegmentationFromNormals%20planar%20segmentation). We created the node since the pcl_ros nodelet was causing some troubles related to time synchronization. 

Anyway, the problem must be somehow related to the PCL. Could you please check if the input cloud is well formed, using RVIZ for instance? The topic is `/stereo_camera/z_filtered_cloud`

beltransen, 2018-01-19:
Closing the issue due to inactivity. If any further help is required, please, open a new issue.

closed 2018-01-19 by beltransen