beltransen/velo2cam_calibration #6: Not enough inliers

agneevguin, 2017-12-12:
I am remapping velodyne_points to laser_pattern/cloud1.
While running the laser_pattern.launch, there is no data published in laser_pattern/centers_cloud and I am getting the following errors:

`Failed to find match for field 'range'.
[ INFO] [1513092899.069292651]: Rot. vector: (0.272492 -0.962158 0.000000) / Angle: 1.598452
[pcl::SampleConsensusModelCircle2D::optimizeModelCoefficients] Not enough inliers found to support a model (3)! Returning the same coefficients.
[pcl::SampleConsensusModelCircle2D::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[ WARN] [1513092899.076805284]: [Laser] Not enough centers: 1
[ INFO] [1513092899.077332401]: [Laser] Processing cloud...
Failed to find match for field 'range'.
[ INFO] [1513092900.124336405]: Rot. vector: (0.263970 -0.964531 0.000000) / Angle: 1.582907
[pcl::SampleConsensusModelCircle2D::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[ WARN] [1513092900.131475392]: [Laser] Not enough centers: 1
[ INFO] [1513092900.131994800]: [Laser] Processing cloud...
Failed to find match for field 'range'.
[ INFO] [1513092901.150450598]: Rot. vector: (0.261055 -0.965324 0.000000) / Angle: 1.595095
[pcl::SampleConsensusModelCircle2D::optimizeModelCoefficients] Not enough inliers found to support a model (3)! Returning the same coefficients.
[pcl::SampleConsensusModel::getSamples] Can not select 0 unique points out of 1!
[pcl::RandomSampleConsensus::computeModel] No samples could be selected!
[pcl::SACSegmentation::segment] Error segmenting the model! No solution found.
[ WARN] [1513092901.152248017]: [Laser] Not enough centers: 1
[ INFO] [1513092901.152711461]: [Laser] Processing cloud...
Failed to find match for field 'range'.
^C[ INFO] [1513092902.114659369]: Rot. vector: (0.270437 -0.962738 0.000000) / Angle: 1.585106
[ WARN] [1513092902.115765398]: [Laser] Not enough centers: 0
`
Am I missing something? I am not sure how to interpret this.

cguindel, 2017-12-13:
Probably you will need to tune some parameters to allow the calibration target segmentation. Please pull the new commit fa6d49d and show the `/laser_pattern/range_filtered_velo` cloud using RVIZ. You should see your calibration target there; otherwise, you need to use dynamic_reconfigure to change the limits of the pass-through filters for the lidar cloud. Nodes whose parameters are to be changed are laser_pattern (passthrough_radius_{min,max}) and pass_through_{x,z}\_velo (filter_limit_{min,max}). Later, you can check `/laser_pattern/pattern_circles` to see if the points belonging to the circles are properly segmented. Yet, don't worry if some frames are skipped due to the lack of inliers. As long as you receive this kind of information messages:

> [Laser] 1/1 frames: 4 pts in cloud

and so on, it should be ok.

agneevguin, 2017-12-14:
This seems to be working well now. Thanks a lot.. :)

closed 2017-12-14 by agneevguin