beltransen/velo2cam_calibration #7: cam_plane_coeffs missing

agneevguin, 2017-12-13:
I am using a ZED camera and there is no separate topic for cam_plane_coeffs (pcl_msgs::ModelCoefficients). Can this be published separately?

cguindel, 2017-12-13:
`cam_plane_coeffs` is remapped to `/planar_segmentation/model` in the stereo_pattern launch file, and `/planar_segmentation/model` is in turn published by the already included v2c_plane_segmentation node, so it should be published regardless of the camera that you are using. You can check if the plane segmentation node is actually publishing the coefficients using `rostopic echo /planar_segmentation/model`.

agneevguin, 2017-12-14:
Thanks a tonne for the code commit. It's much clearer now.
After testing this out, I am getting data in `/stereo_camera/z_filtered_cloud` but not in `/planar_segmentation/model` or `/planar_segmentation/inline`.
This seems like a PCL issue. Do I need to tune the parameters here?

The error is as below:
```
[pcl::SampleConsensusModel::getSamples] Can not select 0 unique points out of 0!
[pcl::RandomSampleConsensus::computeModel] No samples could be selected!
[pcl::SACSegmentation::segment] Error segmenting the model! No solution found.
[ WARN] : Could not estimate a planar model for the given dataset.
```
Can you explain these parameters too:
```
    <param name="~min_line_inliers" value="600"/>
    <param name="~min_plane_normal_z" value="0.8"/>
    <param name="~line_threshold" value="0.02"/>
    <param name="~plane_distance_inliers" value="0.1"/>
    <param name="~min_border_x" value="0.99"/>
```

agneevguin, 2017-12-14:
The error is occurring from the `v2c_plane_segmentation`.

cguindel, 2017-12-14:
`v2c_plane_segmentation` is performing RANSAC to find the most populated plane in the input cloud. The PCL error that you are getting seems to be pointing out that the input cloud has 0 points; i.e., is empty. Make sure (e.g., by using `rostopic echo`) that the **width** field of `/stereo_camera/z_filtered_cloud` is greater than 0; if not, you should tune the passthrough parameters for the stereo cloud as well (similar to #6).  

On the other hand, most of those parameters control the removal of the outer pattern borders. As described in the paper:

> On the other hand, the camera cloud is subjected to a filtering process based on the elimination of lines, since the borders of the calibration target are densely represented in this case. Lines are found using a sample consensus segmentation and selected according to their orientation and the known dimensions of the calibration pattern, to prevent the removal of useful information from the circles themselves. 

`min_line_inliers` is the minimum number of points belonging to a line to be removed, `line_threshold` is the distance threshold for the line RANSAC and `min_border_x` is used to impose that the outer target lines must be roughly parallel to the camera axis. As you can see, the removal process is subject to strong restrictions to avoid missing useful points, but, also from the paper:

> Notwithstanding these considerations, our experiments proved that our segmentation method was largely insensitive to the presence of these borders, except for some very specific poses of the calibration target.  

Finally, `min_plane_normal_z` is used to make sure that we take the correct plane (that one representing the calibration target), by imposing a minimum coefficient for the z component of the plane model. The calibration pattern must be placed in vertical position so it shouldn't change a lot. And `plane_distance_inliers` is the distance (in m) around the plane model in which inliers are considered: [image] in the paper.

shulace13, 2017-12-18:
Hi
I'm having the same trouble, and I'm currently not getting a value greater than 0 in the width if /stereo_camera/z_filtered_cloud . 
So I'm trying to tune the parameters but I'm not sure of where to edit.
Is there some place, other than the real_stereo_pattern.launch file, where I need to edit?

Thank you in advance.

beltransen, 2017-12-18:
Hi @shulace13, for parameter tuning, I suggest you run `rosrun rqt_reconfigure rqt_reconfigure` instead of editing the _.launch_ files. There you can play with the most relevant parameters and see the effects in RVIZ while the node is running. Finally, when you are sure about the correct values you need to set for your specific scenario, you can edit the _.launch_ file for further executions.

shulace13, 2017-12-19:
Hi @beltransen. Thank you very much for your advice. I will try that out!

agneevguin, 2017-12-21:
@cguindel @beltransen I am also using ZED camera for my tests and have tried modifying the parameters up to the best of my understanding. I am still getting the error:
```
[Camera] Could not estimate a circle model for the given dataset.
Not enough centers: 0
[Camera] Estimated plane is not valid. 
```
I am getting `min_plane_normal_z=0` and thus, crashes at `cam_plane_coeffs->values[2]<min_plane_normal_z_`.
Here is a rosbag for reference. https://www.dropbox.com/s/9xkbrwghu1i6oy1/subset_5.bag

The distance from the camera to the board is ~1.8m.
There is a lot of surrounding objects and I tried setting the min and max ranges accordingly. 
Maybe it's a simple parameter but I am unable to figure that out.

Any help would be really grateful.

cguindel, 2018-01-17:
Sorry for the very late reply. That error message indicates that the circle RANSAC was not able to find any circle in the stereo cloud. I have been checking your bag file and I think that, in this specific case, the problem is related to the `camera_info` topics provided by your camera. The Tx parameter of the right camera is set to -0.08189019560813904. According to the [camera_info definition](http://docs.ros.org/api/sensor_msgs/html/msg/CameraInfo.html), this would imply a baseline of 0.00012 m, which seems to be wrong. That is why the resulting point cloud, `/stereo_camera/points2`, does not keep the real dimensions, as you can check using RVIZ, for example. 

Please, let us know if that solves your problem.

agneevguin, 2018-01-23:
I had some issues with the pattern which I need to fix that before I can test this out.
Thanks for testing it and pointing this. I shall confirm once I am able to test it.

agneevguin, 2018-02-20:
Thank you @cguindel It works well now. :)
Closing the issue.

closed 2018-02-20 by agneevguin