beltransen/velo2cam_calibration #8: The time stamp of the tf

shulace13, 2017-12-17:
Hi, I'm having trouble with testing your code with the given .bag file.

The three launch files, laser_pattern, stereo_pattern, and velo2cam_calibration, seems to be working fine.
But when I start up rviz to see what's going on, the camera data isn't showing up.
Every /stereo_camera and /stereo_pattern topic I try to show in rviz, it says the line below
"Message removed because it is too old (frame=[stereo_camera], stamp=[11771.738000000])"

I tried to use this command but that doesn't seem to do the right thing... 
rosparam set use_sim_time true

Many thanks for your help in advance.

cguindel, 2017-12-18:
If you use `rosparam set use_sim_time true`, you will need to run the bag file with the option `--clock`. You can also try to reset RVIZ using the button in the bottom left corner. Either way, our advice is to run the roslaunch files and configure the RVIZ displays before starting to play the rosbag file.

shulace13, 2017-12-19:
I forgot about the `--clock` command! 
It's working out fine now, thank you for the advice.

closed 2017-12-19 by shulace13