beltransen/velo2cam_calibration #9: Using ZED camera

shulace13, 2017-12-19:
Hi, this question may sound dumb but I wanted to be sure.

I'm still having problems with tuning the parameters. `/stereo_camera/z_filtered_cloud` wont't show up like the simulation.
I was wondering if I edited the topic name correctly.
I replaced `/stereo_camera` with `zed`, as you can see in the code below
`<launch>

  <arg name="stdout" default="screen" />

  
  <group ns="zed">
  
    
    <node pkg="nodelet" type="nodelet" name="stereo_manager" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="disparity"
      args="load stereo_image_proc/disparity stereo_manager">
      <param name="disparity_range" value="128"/>
      <param name="correlation_window_size" value="3"/>
      <param name="stereo_algorithm" value="1"/>
      <param name="prefilter_cap" type="int" value="63"/>
      <param name="uniqueness_ratio" type="double" value="10.0"/>
      <param name="speckle_size" type="int" value="100"/>
      <param name="speckle_range" type="int" value="2"/>
      <param name="P1" type="int" value="216"/>
      <param name="P2" type="int" value="864"/>
      <param name="disp12MaxDiff" type="int" value="1"/>
	  <remap from="left/image_rect" to="left/image_rect_color"/>
	  <remap from="right/image_rect" to="right/image_rect_color"/>
    </node>
    <node pkg="nodelet" type="nodelet" name="pointclouder_full"
      args="load stereo_image_proc/point_cloud2 stereo_manager _queue_size:=100">
    </node>
  </group>

  
  <node name="edge_detection" pkg="opencv_apps" type="edge_detection" output="$(arg stdout)" >
    <remap from="image" to="/zed/left/image_rect_color" />
    <param name="use_camera_info" value="false" />
    <param name="debug_view" value="false" />
    <param name="edge_type" value="0" />
    <param name="canny_threshold1" value="100" />
    <param name="canny_threshold2" value="200" />
    <param name="apertureSize" value="3" />
    <param name="apply_blur_pre" value="true" />
    <param name="postBlurSize" value="13" />
    <param name="postBlurSigma" value="3.2" />
    <param name="apply_blur_post" value="false" />
    <param name="L2gradient" value="false" />
  </node>

  
  <node name="v2c_disp_masker" pkg="velo2cam_calibration" type="v2c_disp_masker" output="$(arg stdout)" >
    <remap from="~image" to="zed/disparity" />
    <remap from="~mask" to="edge_detection/image" />
    <remap from="~output" to="zed/edges_disparity" />
    <param name="edges_threshold" value="32" />
  </node>

  
  <group ns="zed">
    
    <node pkg="nodelet" type="nodelet" name="pointclouder_manager_edges" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="pointclouder_edges"
      args="load stereo_image_proc/point_cloud2 pointclouder_manager_edges _queue_size:=100"
      respawn="true">
      <remap from="disparity" to="edges_disparity"/>
      <remap from="points2" to="edge_points2"/>
    </node>
  </group>

  
  <node pkg="nodelet" type="nodelet" name="stereo_pcl_manager" args="manager" output="$(arg stdout)"/>

  
  <node pkg="nodelet" type="nodelet" name="edges_pass_through_z" args="load pcl/PassThrough stereo_pcl_manager" output="$(arg stdout)">
    <remap from="~input" to="/zed/edge_points2" />
    <remap from="~output" to="/zed/edge_z_filtered_cloud" />
    <rosparam>
      filter_field_name: z
      filter_limit_min: 2
      filter_limit_max: 4
      filter_limit_negative: False
      max_queue_size: 1
      keep_organized: False
    </rosparam>
  </node>

  
  <node pkg="nodelet" type="nodelet" name="full_pass_through_z" args="load pcl/PassThrough stereo_pcl_manager" output="$(arg stdout)">
    <remap from="~input" to="/zed/points2" />
    <remap from="~output" to="/zed/z_filtered_cloud" />
    <rosparam>
      filter_field_name: z
      filter_limit_min: 2
      filter_limit_max: 4
      filter_limit_negative: False
      max_queue_size: 1
      keep_organized: False
    </rosparam>
  </node>

  
  <node pkg="velo2cam_calibration" type="v2c_plane_segmentation" name="planar_segmentation" output="$(arg stdout)" required="true">
    <remap from="~input"   to="/zed/z_filtered_cloud" />
    <rosparam>
      segmentation_type: 1
      axis: [0.0, 1.0, 0.0]
      threshold: 0.01
      eps_angle: 0.6
    </rosparam>
  </node>

  
  <node pkg="nodelet" type="nodelet" name="extract_plane_indices" args="load pcl/ExtractIndices stereo_pcl_manager" output="screen">
    <remap from="~input"   to="/zed/z_filtered_cloud" />
    <remap from="~indices" to="/planar_segmentation/inliers" />
    <rosparam>
      negative: false
    </rosparam>
  </node>

  
	<node pkg="velo2cam_calibration" type="stereo_pattern" name="stereo_pattern" output="screen">
	<remap from="~cloud2" to="/zed/edge_z_filtered_cloud"/>
    <remap from="~cam_plane_coeffs" to="/planar_segmentation/model"/>
	<param name="~min_line_inliers" value="200"/>
    <param name="~min_plane_normal_z" value="0.8"/>
    <param name="~line_threshold" value="0.1"/>
    <param name="~plane_distance_inliers" value="0.1"/>
    <param name="~min_distance_between_borders_x" value="0.7"/>
    <param name="~min_distance_between_borders_y" value="1.0"/>
    <param name="~min_line_inliers" value="900"/>
	</node>

  
  <node pkg="nodelet" type="nodelet" name="extract_circle_indices" args="load pcl/ExtractIndices stereo_pcl_manager" output="screen">
    <remap from="~input"   to="/velo2cam_calibration/z_filtered_cloud" />-
    <remap from="~indices" to="/velo2cam_calibration/inliers" />
    <rosparam>
      negative: false
    </rosparam>
  </node>

  
  
  <node pkg="tf" type="static_transform_publisher" name="stereo_ros_tf" args="0 0 0 -1.57079632679 0 -1.57079632679 zed zed_left_camera 100"/>
</launch>`

I edited the code of ZED's SDK so that it publishes `/stereo_camera/*` instead of `/zed/*` but that doesn't seem to do the thing either...

Thank you in advance.

beltransen, 2017-12-19:
Hi @shulace13. if publisher/subscribers are pointing to the required topics you shouldn't need to change the ZED driver. To that end, you can adjust the topics by using the **<remap>** tags inside **<node>** in the _.launch_ file.

However, if you're not sure about if everything is connected properly, you can run `rosrun rqt_graph rqt_graph` and check which links are broken in your nodes graph.

shulace13, 2017-12-20:
Hi @beltransen.
I see. I'll then stick to just changing the tags in the _.launch_ file.

I also checked the connection via rqt_graph, and everything seems to be fine...
Here's the image of the rqt_graph when running `real_stereo_pattern.launch` and `velo2cam_calibration.launch`. I had to run the second one too, just so that the stereo_pattern node would show up.
Does this mean I only have to adjust the parameters?

[image]

shulace13, 2017-12-20:
Hi
Since `/points2` won't show up correctly, I assumed that I needed to tune the `stereo_manager` parameters.
So I'm going to adjust them according to the link below.
http://wiki.ros.org/stereo_image_proc/Tutorials/ChoosingGoodStereoParameters
I've never used this program before, and I think everything will turn out well once I get this straight.

I will comment again after finishing all of this!

shulace13, 2017-12-20:
I've managed to set my own parameters for stereo_image_proc but still can't work it out.
I've noticed that in `real_stereo_pattern.launch` it subscribes to `*/image_rect` but when I run stereo_image_proc alone, it says it's subscribing to `*/image_raw`... 
I'm getting very confused here and I'm very sorry for your inconvenience but any further help will be great...

shulace13, 2018-02-15:
I'm very sorry for the lack of updates.

I came to a conclusion that I needed to make a better calibration board. It seems that I couldn't get the disparity image quite well with it. It also may be the fact that I haven't configured the ZED parameters yet, but that doesn't seem to be a big problem.

So I believe one can calibrate with your system quite well even if they are using the ZED. I'll close this issue for now but if I ever get things straight, I will post something up again.
Thank you very much for the advice!

closed 2018-02-15 by shulace13