beltransen/velo2cam_calibration #11: process of 'laser_pattern' died for 64-layers

syn1993, 2018-03-19:


beltransen, 2018-03-19:
Hi @syn1993 
As I understand you are using the _velo2cam_gazebo_ simulations to feed the _laser_patter_ node.

We haven't experienced any error with the node related to the number of layers of the LIDAR, so we are not quite sure about what can be the cause of the error with that little information. However, based on our experience, I would recommend you to record a .bag file of the LIDAR data when using working with 32 o 64 layer sensors, as the simulator may crash when it cannot provide sensor data at 1Hz due to the high workload (even so, the node is not usually affected)

Regarding the warning saying **Failed to find match for field 'range'.**, it is thrown by PCL libraries (for debug purposes) and does not affect the node execution, so it can be ignored.

Please, try to record the LIDAR data from the simulator and keep us informed of any new finding.

syn1993, 2018-03-24:
Thank you very much for your quick reply. I have tried to record the data from the simulator but still failed with the same message. Then I tried to debug the laser_patter node with GDB and got this message when the error occurred:

[image]

As I'm new to Linux and ROS, I don't know if the message makes sense. I will still look for the reason and shall the feedback if I solved the problem.

syn1993, 2018-03-26:
I have found the reason why it crashed. I forgot to change "RINGS_COUNT" to 64 in "velo2cam_utils.h". A stupid mistake...

closed 2018-03-26 by syn1993