beltransen/velo2cam_calibration #13: fisheye camera pcl_coloring fail

dbsgws, 2018-09-30:
Hi,I`m learning velodyne fuse camera in gazebo .

The stereo-camera`s pointcloud coloring works well !

So I try to build a fish-eye camera model to replace stereo-camera in file
and publish correct frame between lidar and camera .

But,there seems something wrong in coloring (haven`t done calibration in fish-eye camera)
[image]

Coloring function only catch the center of image`s point color .
I wonder if the sensor-type doesn`t fit opencv`s function.
Here is my fish-eye camera sdf .
`<?xml version='1.0'?>
<sdf version='1.6'>
    <model name='fpv_cam'>
    <pose>-0.158979 -0.04405 0.045074 0 -0 0</pose>
      <link name='link'>
        <inertial>
          <mass>0.015</mass>
          <inertia>
            <ixx>4.15e-6</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>2.407e-6</iyy>
          <iyz>0</iyz>
          <izz>2.407e-6</izz>
          </inertia>
        </inertial>
        <collision name='collision'>
          <geometry>
            <box>
              <size>0.01 0.01 0.01</size>
            </box>
          </geometry>
          <max_contacts>10</max_contacts>
          <surface>
            <contact>
              <ode/>
            </contact>
            <bounce/>
            <friction>
              <ode/>
            </friction>
          </surface>
        </collision>
        <visual name='visual'>
          <geometry>
            <box>
              <size>0.01 0.01 0.01</size>
            </box>
          </geometry>
        </visual>
 <sensor name="stereo_camera_1" type="wideanglecamera">
        <camera name='__default__'>
          <horizontal_fov>3.1415</horizontal_fov>
          <image>
            <width>640</width>
            <height>480</height>
          </image>
          <clip>
            <near>0.2</near>
            <far>100</far>
          </clip>
          <lens>
            <type>stereographic</type>
        <custom_function>
        <c1>1.05</c1>   
        <c2>4</c2>      
        <f>1.0</f>      
        <fun>tan</fun>  
      </custom_function>
            <scale_to_hfov>true</scale_to_hfov>
            <cutoff_angle>3.1415</cutoff_angle>
            <env_texture_size>512</env_texture_size>
          </lens>
        </camera>
          <plugin name='camera_plugin' filename='libgazebo_ros_camera.so'>
            <alwaysOn>true</alwaysOn>
            <updateRate>30.0</updateRate>
            <cameraName>stereo_camera_1</cameraName>
            <imageTopicName>image_rect_color</imageTopicName>
            <cameraInfoTopicName>camera_info</cameraInfoTopicName>
            <frameName>stereo_camera_1</frameName>
            <hackBaseline>0.12</hackBaseline>
            <distortionK1>0.0</distortionK1>
            <distortionK2>0.0</distortionK2>
            <distortionK3>0.0</distortionK3>
            <distortionT1>0.0</distortionT1>
            <distortionT2>0.0</distortionT2>
          </plugin>
       </sensor>
        <self_collide>0</self_collide>
        <kinematic>0</kinematic>
      </link>
  </model>
</sdf>`

Hope you can give me some guidance.

beltransen, 2018-11-14:
Hi @dbsgws, sorry for the late response. We have never worked with fisheye cameras, neither in real life nor in Gazebo simulator. Moreover, we haven't tested our calibration method with this kind of devices. Nonetheless, we think the colouring node should work fine if you are able to provide the proper camera-lidar TF and the correct intrinsic parameters of the camera. Good luck!

closed 2018-11-14 by beltransen

dbsgws, 2018-11-22:
thank you very much!
I have used multi-camera to colorize panorama lidar cloud

crankler, 2020-05-22:
@dbsgws can you share your experiment for the fisheye ? Thank you!