beltransen/velo2cam_calibration #15: Failed to find match for field 'range'.

pirobot, 2018-12-19:
Thank you for making this awesome package available.  However, we are having the following issue when running the full_calibration.launch file both in the sample Gazebo world with a VLP-16 and with a real VLP-16 and a Zed stereo camera.  Our environment is Ubuntu 16.04 and ROS Kinetic.

For example, when using the Gazebo world and after launching full_calibration.launch, we get the following stream of messages on the terminal.  The "failed to find match for field 'range'" messages in bold below actually appear yellow on the terminal.  Are these errors fatal or just a warning?

Thanks!

**Failed to find match for field 'range'.**
[ INFO] [1545262150.934490484, 11825.628000000]: [Laser] 530/541 frames: 2120 pts in cloud
[ INFO] [1545262150.976196319, 11825.671000000]: [Camera] 111/111 frames: 444 pts in cloud
**Failed to find match for field 'range'.*
[ INFO] [1545262151.057334826, 11825.727000000]: [Laser] 531/542 frames: 2124 pts in cloud
**Failed to find match for field 'range'.**
[ INFO] [1545262151.198660702, 11825.835000000]: [Laser] 532/543 frames: 2128 pts in cloud
[ INFO] [1545262151.240999520, 11825.879000000]: [V2C] Calibration result:
[ INFO] [1545262151.241041113, 11825.879000000]: x=-0.7959 y=-0.0916 z=0.3912
[ INFO] [1545262151.241060080, 11825.879000000]: roll=0.0001, pitch=-0.0022, yaw=-0.0040
**Failed to find match for field 'range'.**
[ INFO] [1545262151.349309228, 11825.947000000]: [Laser] 533/544 frames: 2132 pts in cloud
**Failed to find match for field 'range'.**
[ INFO] [1545262151.464140880, 11826.034000000]: [Laser] 534/545 frames: 2136 pts in cloud
[ INFO] [1545262151.570089807, 11826.114000000]: [Camera] 112/112 frames: 448 pts in cloud
etc.

cguindel, 2018-12-20:
Hi @pirobot,
You can safely ignore those messages. They are produced internally by PCL. We wish we could remove them but didn't find a simple way to do it.

closed 2018-12-20 by cguindel

pirobot, 2018-12-20:
OK, thanks!