beltransen/velo2cam_calibration #18: Pcl Projection node is slow

BmlmnnsDev, 2019-02-21:
I am using the pcl_projection node in order to validate my results and I expierence a very poor computational performance. I measured the time that is needed to finish the loop which projects each point into the image plane and the whole loop needs about 7 seconds.

```C++
  #pragma omp parallel for
  for (pcl::PointCloud<Velodyne::Point>::iterator pt = trans_cloud->points.begin(); pt < trans_cloud->points.end(); pt++)
  {
    cv::Point3d pt_cv((*pt).x, (*pt).y, (*pt).z);
    cv::Point2d uv;
    uv = cam_model_.project3dToPixel(pt_cv);

    if((*pt).z>0 && uv.x>0 && uv.x < image.cols && uv.y > 0 && uv.y < image.rows){
      COLOUR c = GetColour(int((*pt).range/20.0*255.0), 0, 255);

      cv::circle(image, uv, 3, cv::Scalar(int(255),int(255),int(255)), -1);

      cv::addWeighted(alpha, 0.5, image, 1, 0, nice);
    }
  }
```

I am using the Velodyne VLP-16. OpenMP is installed. Any ideas whats going wrong ?

beltransen, 2019-02-21:
Hi @BmlmnnsDev, I have to ask you to excuse my colleague @cguindel for this piece of code, he wrote it a long time ago late at night with a beer on the desk while doing the last experiments before a deadline. These were hard times... Now, he is very ashamed of this situation and is currently working on a faster solution (not difficult, though). It should be ready by today or tomorrow :wink:

closed 2019-02-22 by cguindel

cguindel, 2019-02-22:
Oh, the old ghosts from the past always come back... Jokes aside, thank you for the heads up. 
It is definitely faster now (we are getting ~2FPS on KITTI sequences).

BmlmnnsDev, 2019-02-25:
I am deeply humbled to have your support and I am astonished by the performance of the new piece of code.

commit b4e13051 (2019-02-22): Fix #18