beltransen/velo2cam_calibration #19: Melodic compilation error

atyshka, 2019-02-26:
I'm trying to compile this package in ros melodic and running into an error:

`In file included from /usr/include/eigen3/Eigen/Core:419:0,
                 from /usr/include/eigen3/Eigen/StdVector:14,
                 from /usr/include/pcl-1.8/pcl/point_cloud.h:46,
                 from /usr/include/pcl-1.8/pcl/conversions.h:50,
                 from /opt/ros/melodic/include/pcl_conversions/pcl_conversions.h:44,
                 from /home/alex/catkin_ws/src/velo2cam_calibration/src/stereo_pattern.cpp:32:
/usr/include/eigen3/Eigen/src/Core/Product.h: In instantiation of ‘struct Eigen::internal::traits<Eigen::Product<Eigen::Matrix<double, 4, 4>, Eigen::Matrix<float, 4, 1>, 0> >’:
/usr/include/eigen3/Eigen/src/Core/Product.h:115:7:   required from ‘class Eigen::internal::dense_product_base<Eigen::Matrix<double, 4, 4>, Eigen::Matrix<float, 4, 1>, 0, 3>’
/usr/include/eigen3/Eigen/src/Core/Product.h:147:7:   required from ‘class Eigen::ProductImpl<Eigen::Matrix<double, 4, 4>, Eigen::Matrix<float, 4, 1>, 0, Eigen::Dense>’
/usr/include/eigen3/Eigen/src/Core/Product.h:71:7:   required from ‘class Eigen::Product<Eigen::Matrix<double, 4, 4>, Eigen::Matrix<float, 4, 1>, 0>’
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:1390:72:   required from ‘static Eigen::internal::transform_right_product_impl<TransformType, MatrixType, 2, 1>::ResultType Eigen::internal::transform_right_product_impl<TransformType, MatrixType, 2, 1>::run(const TransformType&, const MatrixType&) [with TransformType = Eigen::Transform<double, 3, 2>; MatrixType = Eigen::Map<const Eigen::Matrix<float, 3, 1>, 0, Eigen::Stride<0, 0> >; Eigen::internal::transform_right_product_impl<TransformType, MatrixType, 2, 1>::ResultType = Eigen::Matrix<float, 3, 1>]’
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:442:80:   required from ‘const typename Eigen::internal::transform_right_product_impl<Eigen::Transform<Scalar, Dim, Mode, _Options>, OtherDerived>::ResultType Eigen::Transform<Scalar, Dim, Mode, _Options>::operator*(const Eigen::EigenBase<OtherDerived>&) const [with OtherDerived = Eigen::Map<const Eigen::Matrix<float, 3, 1>, 0, Eigen::Stride<0, 0> >; _Scalar = double; int _Dim = 3; int _Mode = 2; int _Options = 0; typename Eigen::internal::transform_right_product_impl<Eigen::Transform<Scalar, Dim, Mode, _Options>, OtherDerived>::ResultType = Eigen::Matrix<float, 3, 1>]’
/usr/include/pcl-1.8/pcl/common/impl/transforms.hpp:319:37:   required from ‘PointT pcl::transformPoint(const PointT&, const Eigen::Transform<Scalar, 3, 2>&) [with PointT = pcl::PointXYZ; Scalar = double]’
/home/alex/catkin_ws/src/velo2cam_calibration/src/stereo_pattern.cpp:356:89:   required from here
/usr/include/eigen3/Eigen/src/Core/Product.h:29:127: error: no type named ‘ReturnType’ in ‘struct Eigen::ScalarBinaryOpTraits<double, float, Eigen::internal::scalar_product_op<double, float> >’
   typedef typename ScalarBinaryOpTraits<typename traits<LhsCleaned>::Scalar, typename traits<RhsCleaned>::Scalar>::ReturnType Scalar;
                                                                                                                               ^~~~~~
`
This error happens in the call to pcl::transformPoint in stereo_pattern.cpp. Any idea why this is happening?

closed 2019-02-27 by cguindel

atyshka, 2019-02-27:
Thanks for the fix and quick response, I’ll try it tomorrow and see!

cguindel, 2019-02-27:
Thanks for the hint, @atyshka. We don't support melodic officially yet, but I hope 79aab466da27b403ec2170b52098a79dac58dfa8 can solve this problem. Please, let us know if it works for you!

atyshka, 2019-02-27:
Yeah will do! We’re looking at using this for multi-lidar calibration. If all goes well I hope to do a PR.

atyshka, 2019-02-28:
Yes I can confirm that with the switch to float this works just fine in Melodic!

commit 79aab466 (2019-02-27): Fix #19