beltransen/velo2cam_calibration #20: How to fusion lidar and camera?

duchengyao, 2019-03-06:
Great project!
I wonder how to fusion lidar and camera image like https://www.youtube.com/watch?v=b-xE-80y0qM

Could you please tell me how to do that
Thanks

cguindel, 2019-03-06:
Hi @duchengyao,

The video was recorded using [RVIZ](http://wiki.ros.org/rviz). Both point clouds (stereo and lidar) are [PointCloud2](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/PointCloud2.html) topics. In order to represent both clouds in their correct positions (so both can be aligned as in the video), both topics must have a proper value of `header/frame_id`, and the transform between the sensors must be broadcasted to tf using, for instance, a [static_transform_publisher](http://wiki.ros.org/tf#static_transform_publisher) like the one generated by our method once you have finished the calibration: look for the calibrated_tf.launch file in the package's launch folder. Of course, the `frame_id` names must be consistent with the arguments of the static_transform_publisher (in our case, `stereo_camera` and `velodyne`).

Let us know if you need more information.

duchengyao, 2019-03-13:
Thanks for your reply, 
I could see the `camera image` and `lidar point` by using static_transform_publisher!

cguindel, 2019-03-13:
I'm glad everything is fine now. Let's close the issue then.

closed 2019-03-13 by cguindel

Git89, 2021-06-18:
> Thanks for your reply,
> I could see the `camera image` and `lidar point` by using static_transform_publisher!

Hi! Dicheng Yao,

Could you help me to look at the question I proposed here when you have time?
I made a calibration which has exact same size as image shown in velo2cam repository. 

<?xml version="1.0" encoding="utf-8" ?>
<launch>
    <arg name="stdout" default="screen" />
    <node pkg="tf" type="static_transform_publisher" name="sensor1_rot_tf" args="0 0 0 -1.57079632679 0 -1.57079632679 rotated_camera_middle camera_middle 10" />
    <node pkg="tf" type="static_transform_publisher" name="velo2cam_tf" args="0.219707 -0.103788 -0.535224 -1.55362 -0.125061 0.177813 Pandar40P rotated_camera_middle 100" />
</launch>

I implemented an interface to get colored point cloud in rviz and get fused results.
Could you explain why it did static_transform_publisher between rotated_camera_frame_id and camera_frame_id, please? 
 There is two parameters (axis and angle threshold) which I don't know how to adjust when extracting plane model. Could you give me some instructions and explain what is the axis and the angle threshold, please? 
Also, this calibration tool can use more than 4 pairs of 3d-3d correspondences from one position by changing location of calibration board, right?
If you can help me answer these questions, I would appreciate it a lot.
Thanks again.

duchengyao, 2021-06-21:
1.I'm not sure about these two parameters, since I haven't used this tool for a long time. As far as I am concerned, the camera is forward to Z axis,and lidar is forward to x/y axis. So it should apply a transform before calibration.
2. The video captured by the camera is not only the calibration board, but also other content. The threshold of distance / angle is set to filter the position of the calibration board, so that the algorithm is focus to calib board
3. Maybe not.

duchengyao, 2021-06-21:
> > Thanks for your reply,
> > I could see the `camera image` and `lidar point` by using static_transform_publisher!
> 
> Hi! Ducheng Yao,
> 
> Could you help me to look at the questions I proposed here when you have time?
> I made a calibration board which has exact same size as image shown in velo2cam repository.
> 
> I implemented an interface to get colored point cloud in rviz and get fused results.
> Could you explain why it did static_transform_publisher between rotated_camera_frame_id and camera_frame_id, please?
> There are two parameters (axis and angle threshold) which I don't know how to adjust when extracting plane model. Could you give me some instructions and explain what is the axis and the angle threshold, please?
> Also, this calibration tool can use more than 4 pairs of 3d-3d correspondences from one position by changing location of calibration board, right?
> If you can help me answer these questions, I would appreciate it a lot.
> Thanks again.

1.I'm not sure about these two parameters, since I haven't used this tool for a long time. As far as I am concerned, the camera is forward to Z axis,and lidar is forward to x/y axis. So it should apply a transform before calibration.
2. The video captured by the camera is not only the calibration board, but also other content. The threshold of distance / angle is set to filter the position of the calibration board, so that the algorithm is focus to calib board
3. Maybe not.