beltransen/velo2cam_calibration #22: Kinetic make fail

bsridatta, 2019-03-29:
For gcc version 6.5.0 20181026 (Ubuntu 6.5.0-2ubuntu1~16.04) 
Many **error: unable to find numeric literal operator ‘operator""Q’**

/usr/include/boost/math/special_functions/detail/erf_inv.hpp:354:46: 
error: unable to find numeric literal operator ‘operator""Q’
          if(is_value_non_zero(static_cast<T>(BOOST_MATH_BIG_CONSTANT(T, 64, 1e-900))))

For gcc version 4.8.5 (Ubuntu 4.8.5-4ubuntu8~16.04.1) 
c++: error: unrecognized command line option ‘-std=gnu++14’
make[1]: *** [velo2cam_calibration/CMakeFiles/laser_pattern.dir/all] Error 2

Any suggestions to get around this problem?

bsridatta, 2019-03-29:
Found similar issues elsewhere and got it to build fine using

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DBOOST_MATH_DISABLE_FLOAT128")
in the cmakelist

bsridatta, 2019-06-15:
Please let me know if it is a good way to do this. I am not able to get a proper image overlay with the given simulation rosbag

beltransen, 2019-09-09:
Hi @bsridatta, we are not sure which is the problem now. Are you still having any compilation issues?

bsridatta, 2019-09-09:
Hi @beltransen, Thanks for response, we moved to Melodic and it is working fine. I will close the issue

closed 2019-09-09 by bsridatta