beltransen/velo2cam_calibration #23: Could not estimate a circle model for the given dataset

eloibove, 2019-04-08:
Hi, I am trying to calibrate a Velodyne VLP32-C and two monocular cameras working as a stereo pair. I am having problems with the stereo pipeline: for now, I have properly generated a stereo pointcloud and changed the parameters using rqt_reconfigure to filter the Z component and I get the **/stereo_camera/edge_z_filtered_cloud** that you can see in the image below. Also, in red, you can see the **/stereo_pattern/circles_cloud** that should be the points corresponding to circles, but I get nonsense. Also, I get the following message:

[pcl::SampleConsensusModelCircle2D::optimizeModelCoefficients] Not enough inliers found to support a model (0)! Returning the same coefficients.
[ WARN] [1554712011.467895294]: [Camera] Could not estimate a circle model for the given dataset.
[ WARN] [1554712011.467980132]: Not enough centers: 0

[image]

I am a bit confused because with the pointcloud I obtain, I think it should be possible to get the centers of the circles... What do you think? 

I also attach a rosbag with my rectified images and the camera info if you need to try out with my data. Thanks in advance!

**BAG file:** 
https://drive.google.com/file/d/1wHIJSwng5Bkc9JyzjBYYgWcZtVthlPLV/view?usp=sharing

eloibove, 2019-04-10:
Hi everyone!
I recalibrated my stereo pair (corrected the distortion coefficients) and also it turned out that I had the units of the translation matrix from cam1 to cam2 in millimeters... When I switched to meters it finally worked! 
[image]

closed 2019-04-10 by eloibove

290811, 2022-11-10:
hi, may i know which file that include distortion coefficients, right now, i face the same problem that i can't detect the centers of the circles

halops, 2022-11-10:
Hi, it's been a while since this issue and I'm not sure I can help, maybe it's better if you create a new issue.
In our particular case, to generate the stereo pointcloud, we needed to publish `sensor_msgs/CameraInfo` messages containing the camera pair calibration parameters, and that's where our issue was.

290811, 2022-11-10:
already posted it in another account, but didn't get the reply yet.. thank you