beltransen/velo2cam_calibration #24: translation error

dixiAI, 2019-08-16:
Hello!
have someone some experiences with the calibration results?. In the simulation I get big errors in the z axis and/or in the y axis from 2 cm to 10 cm. It have realized they depend of the position of the calibration target or the position between the sensors.
Do you get gut results? I mean calibration errors under 1 cm?
Thank you very much in advance for your help and comments
cheers
Dixie

beltransen, 2019-09-09:
The reported results correspond to very different calibration scenarios. as described in the paper. We performed experiments in a variety of relative positions between the sensors and the calibration target in order to test the robustness of the method against challenging calibration configurations. We have also received feedback from other users that have used this method to calibrate the sensors of their own vehicles with successful results. 

It looks a little bit weird to get these errors while working in a simulated environment. Have you double-checked that the parameters are properly adjusted?

cguindel, 2019-11-11:
We are closing this issue because it has been inactive for two months, and no further information has been provided.

closed 2019-11-11 by cguindel

iyuner, 2021-01-15:
Hello I also met such problem- translation error is big and rotation is fairly good in the simulation. @dixiAI did you solve it in the end?