beltransen/velo2cam_calibration #25: Ros-Kinetic Fail

rukie, 2019-09-20:
During the build I'm seeing the errors below. Am I missing some dependencies? 

Command to install:
```bash
catkin_make --only-pkg-with-deps velo2cam_calibration
```
Errors:
```bash
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
velo2cam_calibration/CMakeFiles/stereo_pattern.dir/build.make:62: recipe for target 'velo2cam_calibration/CMakeFiles/stereo_pattern.dir/src/stereo_pattern.cpp.o' failed
make[2]: *** [velo2cam_calibration/CMakeFiles/stereo_pattern.dir/src/stereo_pattern.cpp.o] Error 4
CMakeFiles/Makefile2:996: recipe for target 'velo2cam_calibration/CMakeFiles/stereo_pattern.dir/all' failed
make[1]: *** [velo2cam_calibration/CMakeFiles/stereo_pattern.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
velo2cam_calibration/CMakeFiles/laser_pattern.dir/build.make:62: recipe for target 'velo2cam_calibration/CMakeFiles/laser_pattern.dir/src/laser_pattern.cpp.o' failed
make[2]: *** [velo2cam_calibration/CMakeFiles/laser_pattern.dir/src/laser_pattern.cpp.o] Error 4
CMakeFiles/Makefile2:1117: recipe for target 'velo2cam_calibration/CMakeFiles/laser_pattern.dir/all' failed
make[1]: *** [velo2cam_calibration/CMakeFiles/laser_pattern.dir/all] Error 2
```

cguindel, 2019-09-23:
We have not run into that error, but I will dare to say that it might be related to a lack of RAM. You could try to reduce the number of jobs started by catkin_make as in [here](https://answers.ros.org/question/59031/how-do-i-limit-the-number-of-cores-used-in-the-build-process-by-catkin/) or to increase the swap size.

rukie, 2019-09-23:
That's super clever of you. At first I thought no way, my machine has 96 gb of RAM... and then I realized I'm on a VM running Ubuntu with 8gb setup. 

Increasing the VM RAM resolved this the issue! Thank you!

What in the error hinted a memory issue to you?

closed 2019-09-23 by rukie

cguindel, 2019-09-23:
Well, I noticed the "c++: internal compiler error: Killed (program cc1plus)" line and did some googling to confirm my suspicions... 

I am glad you could solve your problem. Thank you for sharing the resolution!