beltransen/velo2cam_calibration #27: [pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!

lemon5254, 2020-03-16:
Hello!
Thanks for the package.
when i use real velodyne VLP16, i got some problem!
when i use velodyne VLP16 package,"roslaunch velodyne_pointcloud VLP16_points.launch"

and then "roslaunch velo2cam_calibration laser_pattern.launch"

i got a error in terminal 
"[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!"

how can i slove 
this is my rqt_graph
[image]

and error message
[image]

cguindel, 2020-03-16:
Hi @lemon5254 

You must adjust a series of passthrough filters to fit your setup (i.e., the position of the calibration marker with respect to the lidar). In short, you have to run:

`rosrun rqt_reconfigure rqt_reconfigure`

And then modify the parameters `filter_limit_min` and `filter_limit_max` of the two passthorugh filters, and the `passthorugh_radius_min` and `passthorugh_radius_max` parameters of _laser_pattern_, until:

- `/velodyne/xy_filtered` contains all the points belonging to the calibration marker AND the wall behind.
- `/laser_pattern/range_filtered_velo` contains ONLY the points belonging to the calibration marker.

You can check these point clouds in _rviz_ while tuning the rqt_reconfigure parameters. When you find the proper values, please add them to `laser_pattern.launch` and rerun it. 

Let us know if this fixes your problem.

lemon5254, 2020-03-20:
@cguindel Thank your reply! i had solved this problem!!!

And i have another question

How can you get ground truth in gazebo ?
First, i think that is using (velodyne pose - camera pose) in gazebo
But i can't get same ground truth with setting 8 and setting 9 in your paper

beltransen, 2020-03-24:
Hi @lemon5254, we have never used Gazebo to obtain the pose between sensors, so I'm not able to give you a straightforward solution. However, I've done a quick search on Google and found [this](https://answers.gazebosim.org//question/12212/get-pose-of-a-model-from-a-plugin/), which may be of interest. 

On the other hand, if you know how to get poses from Gazebo, it would be nice if you can share the steps you're following with us so we may be able to help you find a solution.

lemon5254, 2020-03-25:
@beltransen OK!!! Thank your reply!! I will continue to study! thanks!

closed 2020-03-25 by lemon5254

KangYongHa, 2022-04-04:
Hi. I had a question while reading your writing.I know that there are a lot of useless topics in velodyne_nodelet_manager. Can you tell me how you removed them?