beltransen/velo2cam_calibration #30: Information on markers

Wing-ka-king, 2021-03-01:
Hi!

Since I have no background in computer vision, but only controls, I am struggling to decide the markers for target.
And since readme and howto do not provide this information, I'm opening an issue.

I realize that you have used aruco markers, but that's all. What grid size should I use, and what ID should I assign to these markers?

Thanks

Wing-ka-king, 2021-03-03:
So for the future users who do not want to get into defining and identifying new markers and just use the ones you have shown.,its
6x6 grid aruco markers with ID 1,2,3,4. 
`Top Left - ID 1`
`Top Right - ID 2`
`Bottom Left - ID 3`
`Bottom Right - ID 4`
Printed to correct physical dimension using tools provided by https://github.com/okalachev/arucogen.

closed 2021-03-03 by Wing-ka-king

cguindel, 2021-03-11:
Thank you, @Wing-ka-king. Your information is correct.

If for whatever reason, you need to use other dictionary, you can change it here: https://github.com/beltransen/velo2cam_calibration/blob/a26164200a38cb9211f000e7606e400fdfc67ceb/src/mono_qr_pattern.cpp#L522

Also, the IDs are defined here: https://github.com/beltransen/velo2cam_calibration/blob/a26164200a38cb9211f000e7606e400fdfc67ceb/src/mono_qr_pattern.cpp#L201

We chose a pretty standard set of markers to keep things simple, but we think that changing it wouldn't make a great difference.

maciej-3, 2021-03-18:
> https://github.com/beltransen/velo2cam_calibration/blob/a26164200a38cb9211f000e7606e400fdfc67ceb/src/mono_qr_pattern.cpp#L522
> 
> Also, the IDs are defined here:
> 

Hi @cguindel. I probably do not understand here the dimensions. Namely, why the size of the marker is set to be 250 in the code above (`cv::aruco::DICT_6X6_250`) while the size for the aruco tag on the calibration target is 200x200mm?

[image]

Also, if I may: if the size of the aruco target should be 200x200 (including the white margins as in your photo on the left), how wide the margins should be and what is the physical size of the tag then?

beltransen, 2021-03-18:
Hi @maciej-3,
the white margins are not required if the calibration target is white. We do recommend their usage in case the board is made of wood or any other non-white material. The margin size can be any, but it is recommended to keep it small so that the aruco markers are as big as possible. In the case of the figure, the marker side is set to 150mm, centered on the 200x200 space (thus, a white margin of 25mm is used).

Once you have defined the final design, the marker_size parameter inside mono_pattern.launch must be set accordingly using the marker size (without the padding).