beltransen/velo2cam_calibration #31: Dealing with the tf tree

Wing-ka-king, 2021-03-02:
Hi!
Before starting the calibration process using ``velo2cam``, does a transform between sensors have to be defined?
And the `calibrated_tf.launch` file, that is generated at the end of the procedure, does it take into account the existing transforms if any?
For example, we have tf tree defined in the urdf of the robot, having 3 lidars and 1 camera. If I fire up 2 of those lidars and calibrate them, how should I interpret the generated `calibrated_tf.launch` file.

cguindel, 2021-03-11:
Hi @Wing-ka-king. I'm not sure if I understood your question, but I'll try to answer it. The method is designed to provide the TF transformation between the frames that appear in the `frame_id` field of the sensor topics header. The auto-generated `calibrated_tf.launch` provides this transformation, sometimes with some intermediate steps (if any of the sensors is a camera). 

You don't need to broadcast any TF transformation before starting the procedure; actually, if you do, they will be ignored by velo2cam. After the calibration, the static_transform_publisher(s) contained in `calibrated_tf.launch` will publish the TF transform between the two sensors that have been calibrated, and you will need to create the rest of the TF tree separately.

I don't know if this solves your issue. Let us know if you need more information.

Wing-ka-king, 2021-03-15:
Hi @cguindel !
Yeah I did try some trials-and-errors in the mean time and finally found out 

> You don't need to broadcast any TF transformation before starting the procedure 
 
is the way to go.
Thank you! This solves my issue.

closed 2021-03-15 by Wing-ka-king