beltransen/velo2cam_calibration #34: How did you calibrated your camera intrinsically for real-world experiments?

maciej-3, 2021-05-30:
Hi! Thanks for a very nice calibration package.

While using it, however, I have noticed that the intrinsic camera calibration is quite critical for the camera-lidar extrinsic calibration. May I ask what package/software you used for the camera intrinsic matrix and distortion coefficient to obtain the results as shown in your paper? Any tips on that?

cguindel, 2021-06-07:
We usually perform intrinsic calibration through the [camera_calibration](http://wiki.ros.org/camera_calibration) package, which we find pretty nice. However, when working with wide lenses, we often found it necessary to change the default _plumb_bob_ model (with five distortion parameters) to a more complex _rational_polynomial_ model (described by eight parameters). Note that the number of distortion coefficients is an optional argument of the `cameracalibrator.py` script.

Of course, you also must perform the procedure carefully and get a lot of different poses and scales so that all the bars on the calibration GUI are full.

maciej-3, 2021-06-10:
Many thanks for your answer!

closed 2021-06-10 by maciej-3

maciej-3, 2021-07-21:
I have managed to calibrate my camera-lidar suite successfully using your package and the ROS calibrator. Thanks for your answer, @cguindel. However, I could not fully fill in the "size" bar while using the ROS calibrator and I could not find anything about it on the Internet. I used an A0-size 8x6 checkerboard during the calibration. Have you also experienced this problem guys?

cguindel, 2021-07-22:
We have indeed experienced that, but we don't think it is actually an important issue. As far as we have deduced, filling the "Size" bar requires moving the pattern very, very close to the camera, up to the point that the outer squares are no longer visible (but the inner corners are). Anyway, you will probably obtain a good result even if this bar is not totally complete.

cguindel, 2021-09-02:
Yes, our package supports the rational polynomial distortion model. Actually, the distortion parameters are only needed for the OpenCV's aruco functions, which can handle different distortion models seamlessly. However, note that the monocular branch will require the unrectified image (image_color), as explained [here](https://github.com/beltransen/velo2cam_calibration/blob/master/HOWTO.md).

Enes1097, 2021-09-02:
Oh, that's great. Thank you!

I've tried using "image_color", the unrectified color image, but then realized that my camera puts out the images in a BGRA format which is not supported by the OpenCV aruco function. Which is why I'm using the image_mono topic - it seems to be working. Is there any difference if use the mono image instead of the color image?

cguindel, 2021-09-10:
As far as I know, `cv::aruco` functions accept either a grayscale image or a color image; actually, according to the source code, [all input images are converted to grayscale as a first step](https://github.com/opencv/opencv_contrib/blob/78acb196bc5aaa12c935401197869e22657b2f15/modules/aruco/src/aruco.cpp#L992), so I don't think it will cause any problem.