beltransen/velo2cam_calibration #39: Can I do calibration between lidar and mono-camera with v1.0?

konnoeric, 2021-09-07:
Hi. Thank you very much for your excellent contribution.

 You said before that I can color the point cloud by implementing the code in v1.0. Can I also do calibration between lidar and mono-camera with v1.0? My understanding is that the v1.0 is using the image of left camera of stereo camera. But  I only have mono-camera. The description in v1.0 is only telling which launch file to implement to do calibration between lidar and stereo camera. 

Would you tell me which one should I launch to do that? I am asking this because I made the board which has 4 circles last year based on the calibration target details in v1.0. But now the v2.0 is requiring different measurements of the target board and it is a bit hard to create it again. So it will be very helpful if I can do the calibration between lidar and mono-camera (usb camera)  by using v1.0. Or would you describe me which parameter I should change to adapt to the target board which I made based on v1.0? In such case, maybe I can do the calibration with v2.0 by changing the parameter of the target board.

konnoeric, 2021-09-08:
I have launched `roslaunch velo2cam_calibration lidar_pattern.launch cloud_topic:=/velodyne_points` and it said 

> [ WARN] [1631084547.846173751]: [LiDAR] Could not detect pattern edges.
So is this because I was using the target board in v1.0 measurement?

I looked at the cfg file to see if there is any parameter to be modified but what I only found that might be representing the size of the target board is only `gen.add("circle_radius", double_t, 0, "Radius of pattern's circles", 0.12, 0, 1)`. Is the Lidar.cfg only giving the radius of circles? Is lidar calibration only considering the size of circles?

cguindel, 2021-09-09:
Hi @konnoeric,

The v1.0 codebase does not include the monocular branch. As you said, it was designed to work with our previous iteration of the calibration pattern, which didn't include ArUco markers. So, if you want to use the pcl_projection and pcl_coloring nodes, you should add (and maybe adapt) these nodes to the v2.0 code.

On the other hand, we changed the calibration pattern because the former was prone to originate false circle detections at the center of the board. The new version has more separation between holes to avoid that.

You can still use our previous version of the pattern with the v2.0 with two caveats: first, if you want to calibrate a monocular camera, you will naturally need to include the ArUco markers; and second, you need to modify some parameters, as you correctly guessed. Apart from `circle_radius` (which shouldn't change in this case), you need to modify the node parameters `delta_width_circles` and `delta_height_circles`, which contain the x and y distance between holes. For the monocular node, you will also need to set `marker_size`, `delta_width_qr_center` and `delta_height_qr_center`. The first is pretty self-explanatory, and the other two represent the distance from the center of each marker to the center of the target. I hope I do not forget any others.

closed 2021-12-23 by cguindel