beltransen/velo2cam_calibration #42: <------ Request------->

THARUN-V, 2021-11-11:
Hi, can u provide bag file to check how the calibration algo works.
if possible can u please make a detailed step by step youtube video on how to use this package.

cguindel, 2021-12-23:
I am sorry, but we are unable to fulfill your request. We made an effort to release this code to the open-source community, but our professional duties prevent us from offering personalized support to users. Please read our [paper](https://arxiv.org/abs/2101.04431) and [HOWTO guide](https://github.com/beltransen/velo2cam_calibration/blob/master/HOWTO.md) carefully, and feel free to check and modify the code to suit your needs. We appreciate your understanding.

closed 2021-12-23 by cguindel

THARUN-V, 2022-03-06:
Thanks for replying , but my problem is i am not able to proceed further , i did as per the documentation , i have some doubts to clarify
--> whether there need to be a flat wall behind the target as per the documentation
--> or is it ok to keep target in front of un structured environment

when i run velo2cam with my sensor data the target is not getting detected.

cguindel, 2022-03-18:
The function of the "wall" that must be placed behind the pattern is to get measurements from the lidar beams that go through the holes, so that circle edges can be effectively extracted taking into account the depth difference between foreground (pattern) and background (circles) points. You can tune the minimum space (in meters) required between the pattern and the background with the _gradient_threshold_ parameter here:
https://github.com/beltransen/velo2cam_calibration/blob/a26164200a38cb9211f000e7606e400fdfc67ceb/src/lidar_pattern.cpp#L523

Therefore, the background does not need to be a perfect plane, although it should provide reasonably good lidar reflections so that the difference between both sets of points is clear and the separation is unambiguous.