beltransen/velo2cam_calibration #44: [LiDAR] Unable to find a candidate set that matches target's

msqiu, 2022-03-15:
Hi, I am calibrating a ToF camera and a RGB camera. The generated point cloud is filtered as follows:

[image]

From the viewer the circles are very clear but it shows the warning: `[LiDAR] Unable to find a candidate set that matches target's` like this:
[image]

So do you know the reason?

The `rqt_configure `parameters:
`!!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  angle_threshold: 0.4241150082333
  centroid_distance_max: 1.0
  centroid_distance_min: 0.0
  circle_radius: 0.12
  groups: !!python/object/new:dynamic_reconfigure.encoding.Config
    dictitems:
      angle_threshold: 0.4241150082333
      centroid_distance_max: 1.0
      centroid_distance_min: 0.0
      circle_radius: 0.12
      groups: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      id: 0
      name: Default
      parameters: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      parent: 0
      passthrough_radius_max: 2.0
      passthrough_radius_min: 0.4
      state: true
      type: ''
      x: 0.0
      y: 0.0
      z: 1.0
    state: []
  passthrough_radius_max: 2.0
  passthrough_radius_min: 0.4
  x: 0.0
  y: 0.0
  z: 1.0
state: []
`

aseligmann, 2022-03-17:
I am also experiencing the `[LiDAR] Unable to find a candidate set that matches target's geometry` issue.
Have you managed to find a solution?

@cguindel Do you have any recommendations for how to proceed?

cguindel, 2022-03-18:
I will do my best to help you both, but I am afraid I do not have enough information to get a complete picture of what is happening in your particular cases. 

@msqiu, as you are probably aware, we haven't tested our method on ToF cameras. I can see you are using the _lidar_pattern_ extractor; how are you dealing with [the 'ring' field in the pointcloud](https://github.com/beltransen/velo2cam_calibration/blob/bfba57837e1efd979f8514512756b450f921e2aa/include/velo2cam_utils.h#L62
)? We expect the data to be structured according to scan planes and I am not sure if this is the case for you.

@msqiu and @aseligmann, The `[LiDAR] Unable to find a candidate set that matches target's geometry` message can imply that the parameters expressing the dimensions of the pattern are not correctly set, or that the circle segmentation procedure is extracting outliers that indeed do not match the pattern geometry. 

I would recommend rebuilding the package with the DEBUG flag set here set to 1: https://github.com/beltransen/velo2cam_calibration/blob/bfba57837e1efd979f8514512756b450f921e2aa/include/velo2cam_utils.h#L30 

Then, you can visualize the point cloud _laser_pattern/pattern_circles_. In there, you should clearly distinguish the contour of the pattern (i.e., all the points belonging to the edges); if not, then you should probably adjust your pass-through filters more tightly to avoid outliers during plane fitting. Besides, you must make sure that each circle is represented by at least two LiDAR planes. 

Maybe you can try this and report back.

talhasu, 2022-04-14:
I had the same error as @aseligmann. i tried the steps you recommended but without any success. following is the screenshot of the error I am getting. I have a question. How accurate should the pattern be? Mine is differing by 1 cm in some cases. Do you think that is causing the error?
[image]

cguindel, 2022-04-19:
I think 1 cm of deviation shouldn't prevent you from obtaining a result because the tolerance is set to 6 cm [here](https://github.com/beltransen/velo2cam_calibration/blob/bfba57837e1efd979f8514512756b450f921e2aa/include/velo2cam_utils.h#L37). Still, it might affect the accuracy of the calibration, so you probably want to adjust some parameters such as _delta_width_circles_, _delta_height_circles_, and _circle_radius_ in the _lidar_pattern_ node to make them fit your setup. If that doesn't solve your problem, you will need to check the intermediate clouds generated within the procedure (such as the _laser_pattern/pattern_circles_ topic I mentioned in my previous reply) in RVIZ to see where the problem is.

cguindel, 2022-08-31:
Closing the issue because it has not had recent activity. Thank you for your contributions.

closed 2022-08-31 by cguindel

bclaw524, 2022-10-14:
Hi, 

I am having a similar issue with the lidar calibration. Where it is unable to find a candidate set that matches the target's geometry. 

I'm using a velodyne VLP-16, and I have managed to isolate the zyx and range filter cloud as stated in the directions, however I am still having this issue.

Enabling debug mode produces this output:

[image]

I have managed to get it to detect 4 centers, as below, but it still gives me the same issue. 

[image]

I was also able to produce a ring-field point cloud. 

[image]

Any insight on how I can proceed on this issue?