beltransen/velo2cam_calibration #45: Either "Not enough centers" or "Could not estimate a planar model for the given dataset"

kolja183, 2022-05-05:
Hi, I am fairly new to ROS but I dont know how to proceed from here or what my mistakes are. 

Setup:
The lidar and the camera stand 1.5m from the board. Between board and door there is another 0.5m. 
[image]
[image]
[image]

camera: [ids gv 5250](https://de.ids-imaging.com/store/gv-5250se.html)
lidar: [ouster os1](https://ouster.com/products/scanning-lidar/os1-sensor/)

The camera is monocular. Camera and lidar collect data at 10Hz. I wrote the camera publisher partly myself. Looking like this:

`
    ros::NodeHandle nh;
    image_transport::ImageTransport it(nh);
    image_transport::Publisher pub = it.advertise("camera/image", 1); 

    ros::Publisher pub_camera_info = nh.advertise<sensor_msgs::CameraInfo>("camera/camera_info", 1);
    sensor_msgs::CameraInfo camera_info;

    while (m_running)
    {
        try
        {
            // Get buffer from device's datastream
            const auto buffer = m_dataStream->WaitForFinishedBuffer(5000);

            QImage qImage(m_imageWidth, m_imageHeight, QImage::Format_RGB32);

            // Create IDS peak IPL image for debayering and convert it to RGBa8 format
            const auto image = peak::BufferTo<peak::ipl::Image>(buffer).ConvertTo(
                peak::ipl::PixelFormatName::BGRa8, qImage.bits(), static_cast<size_t>(qImage.byteCount()));

            
            cv::Mat cvimage(qImage.height(),qImage.width(),CV_8UC4, (uchar*)qImage.bits(),qImage.bytesPerLine()); //cv image auf qimage erstellen
            sensor_msgs::ImagePtr msg = cv_bridge::CvImage(std_msgs::Header(), "bgra8", cvimage).toImageMsg(); //sensor image msg erstellen

            (*msg).header.frame_id = "camera"; //std::to_string(m_frameCounter);

            pub.publish(msg); //msg publishen

            
            camera_info.header.stamp = ros::Time::now();
            camera_info.header.frame_id = "camera";

            camera_info.width = m_nodemapRemoteDevice->FindNode<peak::core::nodes::IntegerNode>("Width")->Value();
            camera_info.height = m_nodemapRemoteDevice->FindNode<peak::core::nodes::IntegerNode>("Height")->Value();

            camera_info.distortion_model = "plumb_bob";

            pub_camera_info.publish(camera_info);

            // Queue buffer so that it can be used again
            m_dataStream->QueueBuffer(buffer);

            // Emit signal that the image is ready to be displayed
            emit imageReceived(qImage);

            m_frameCounter++;
        }
`

So i have an topic camera/image and camera/camera_info which should be the required topics. I read a lot about the frame_id of the headers but i still dont fully understand it. Especially why i need them in this context, since camera_info and the camera/image descripe the same images and thus dont need different coordination system. Anyways, here would be **my first question** if I defined the frame_id correct or if I made a mistake here allready.  The header of my lidar data has no frame_id set. Do i need that in this case? Since there is no argument in the launch file i thought i dont need it. 

The point cloud i am publishing in my lidar consist of the following point which is similar to the Velodyne Point:

struct EIGEN_ALIGN16 Point_xyzir{  
	PCL_ADD_POINT4D;  
	float intensity;  
	uint16_t ring;  
	float range;  
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
};

I follow the manual step by step and the launch files work. It says in the manual "For monocular devices, the only relevant data is the image, where the pattern should be fully visible. For range data, the checklist depends on the modality.". So my **second Question** would be if I understand correctly, that i dont have to check my camera image in rviz since i use a monocular device. I tried to check the image in rviz but i get error messages because of my "camera" frame_id that i dont publish anywhere. I get the error: "CameraInfo/P resulted in an invalid position calculation (nans or infs)". I tried to resolve it but it didnt work and i was hoping that this was not the problem bcs of the sentence in the manual mentioned above.
[image]

Next i try to calibrate the lidar with rviz. The two topics "lidar_pattern_0/zyx_filtered" and "lidar_pattern_0/range_filtered_cloud" that I receive look like this on their own:
[image]
[image]

 There would be my **third question**: Should they overlap like this? In the manual the datapoints seem to not be overlapping. My topics seem to have identical points. Am i doing something wrong here?

I manage to be able to cut out the calibration pattern out pretty good with the following settings:
[image]

With this setting the terminal with the lidar launch node outputs following warnings:
[ WARN] [1651764290.766055097]: [LiDAR] Not enough centers: 2
[ WARN] [1651764290.867160361]: [LiDAR] Not enough centers: 3
[ WARN] [1651764290.967275613]: [LiDAR] Unable to find a candidate set that matches target's geometry
[ WARN] [1651764291.066757160]: [LiDAR] Not enough centers: 0
[ WARN] [1651764291.166676958]: [LiDAR] Not enough centers: 0
[ WARN] [1651764291.265684512]: [LiDAR] Not enough centers: 0
[ WARN] [1651764291.366198506]: [LiDAR] Not enough centers: 0
[ WARN] [1651764291.468210932]: [LiDAR] Unable to find a candidate set that matches target's geometry
[ WARN] [1651764291.566586082]: [LiDAR] Not enough centers: 0
[ WARN] [1651764291.666599987]: [LiDAR] Not enough centers: 0
[ WARN] [1651764291.771358378]: [LiDAR] Not enough centers: 1

After configuring the second topic it looks like this:
[image]

This seems correct, but the launch node now says "Could not estimate a planar model for the given dataset" and i have no idea what i am doing wrong. 

When i build the workspace i have no errors but i get warnings regarding opencv:

[ 88%] Linking CXX executable /home/hauke/catkin_ws/devel/lib/ouster_ros/img_node  \n
[ 88%] Built target img_node  
[ 89%] Linking CXX executable /home/hauke/catkin_ws/devel/lib/velo2cam_calibration/velo2cam_calibration  
[ 89%] Built target velo2cam_calibration  
[ 91%] Linking CXX executable /home/hauke/catkin_ws/devel/lib/velo2cam_calibration/mono_qr_pattern  
/usr/bin/ld: Warnung: libopencv_imgcodecs.so.4.2, benötigt von /opt/ros/noetic/lib/libcv_bridge.so, kann in Konflikt geraten mit libopencv_imgcodecs.so.405
/usr/bin/ld: Warnung: libopencv_features2d.so.4.2, benötigt von /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.4.2.0, kann in Konflikt geraten mit libopencv_features2d.so.405
/usr/bin/ld: Warnung: libopencv_imgproc.so.405, benötigt von /usr/local/lib/libopencv_imgcodecs.so.4.5.5, kann in Konflikt geraten mit libopencv_imgproc.so.4.2
/usr/bin/ld: Warnung: libopencv_core.so.405, benötigt von /usr/local/lib/libopencv_imgcodecs.so.4.5.5, kann in Konflikt geraten mit libopencv_core.so.4.2
[ 91%] Built target mono_qr_pattern
[ 92%] Linking CXX executable /home/hauke/catkin_ws/devel/lib/velo2cam_calibration/stereo_pattern
[ 92%] Built target stereo_pattern
Scanning dependencies of target os_cloud_node
Scanning dependencies of target os_node
[ 94%] Building CXX object ouster_example-master/ouster_ros/CMakeFiles/os_node.dir/src/os_node.cpp.o
[ 95%] Building CXX object ouster_example-master/ouster_ros/CMakeFiles/os_cloud_node.dir/src/os_cloud_node.cpp.o

(sorry for formatting)
As i said i am new to ROS and to C++ and cmake aswell. Could this be the problem maybe?

If I can provide any other information just let me know. I would appreciate any help, 
thank you.

kolja183, 2022-05-13:
Yes, i have the same exact dimensions. I can try to set `centroid_distance_min` higher. I wont be able to test it for the next 6 Days i think. It sill seems suspicious to me that `lidar_pattern_/zyx_filtered` and `lidar_pattern_/range_filtered_cloud ` have the exact same points in them thought.

cguindel, 2022-05-16:
Hi @kolja183,

Here we go with your questions:
1. The _frame_id_ field is used by the ROS' tf system to relate different sensors to each other. Basically, each sensor is expected to have a unique _frame_id_ string that represents the position of the sensor. Therefore, the image and camera_info topics should have the same _frame_id_. You will also need a header with a filled _frame_id_ field in your LiDAR data; we don't ask for it in the launch file because it's obtained automatically from the point cloud topic. Note that, according to your screenshots, your LiDAR data is already using the _os_lidar_ frame id so this shouldn't be a problem.
2. You should be able to show the images from your cameras in RVIZ using an _Image_ display. I can't see your picture, but checking your camera driver and the error message, I suspect you are trying to use the camera without obtaining its intrinsic parameters before, which is a requisite for our method. Please check [this](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) to learn how to perform the camera calibration and obtain the _image_rect_ or _image_rect_color_ topics that you should be using as input in our package.
3. The setup seems ok, but you must follow our how-to to adjust the passthrough filters. If _lidar_pattern_0/zyx_filtered_ and _lidar_pattern_0/range_filtered_cloud_ are the same, it is likely because the radial passthrough filter is so loose that it is not removing any point. You must tune the _passthrough_radius_min_ and _passthrough_radius_max_ parameters in the _lidar_pattern_0_ node before, as explained in our manual. On the other hand, in your final configuration, the _range_filtered_cloud_ looks as expected, but the _zyx_filtered_cloud_ should contain **both** the calibration pattern and the wall behind, not only the wall behind as in your screenshots.

Regarding the OpenCV conflict warnings, they are outside the scope of our package, but I suspect you have two different versions of OpenCV installed on your system.

I hope this helps you.

kolja183, 2022-05-16:
Thank you for your answer. I am ill atm so i wont be able to test and try your tips until next week. I will update and close the issue if I am able to fix my problem.

unnikrishnanrnair, 2022-06-08:
Hi, I am stuck in a similar issue.... 
This is my range filtered pointcloud.
[image]
This is my xyz filtered pointcloud.
[image]
I have updated the number of channels to 128 in lidar_pattern.cpp and also I have ensured the board is of the correct dimensions. But still I get the warning "Not enough centers: x" where x ranges from 0-3.

kolja183, 2022-06-08:
My problem was fixed @cguindel. One problem was as you stated my intrinsic camera parameters. I followed your links which worked perfectly well. The other problem I had was that my camera driver did not publish a camera_info topic and I wrote the code doing that myself. Turns out the code that I wrote (see my post) did not work as intended. When I am at work again i can show what code I used to get the info_topic workingn show what code I used to get the camera_info topic working. 

I think you need to give some more information so someone can help you @unnikrishnanrnair. Does the setup find all your camera images? If not I would work on that first. Try to experiment with the angle of the board and the distance from board to the wall, that helped in my calibration case at least.

jasonCastano, 2022-06-23:
Hi, could someone tell me where I can find the pattern for the camera?

lbouz102, 2022-08-09:
@unnikrishnanrnair Did you end up solving this issue? My issue is #51 and I'm curious if you had a sparse pattern cloud like mine.

unnikrishnanrnair, 2022-08-10:
@lbouz102 I have got it working for now. For a single position (not 3 as the package suggests) we are getting the following results. Setting things to debug and verifying that we have the cumulative pointclouds from lidar_pattern and mono_pattern being generated helped us to verify it. But still the quality of the parameters seem not so good (looking at the projection of the lidar points into the image). Dont know if this projection has to be done onto the raw image or the undistorted image.
[image]
[image]
[image]

cguindel, 2022-08-31:
If the camera_info has the distortion parameters, the calibration is computed for the undistorted (_rect) image. Undoubtedly, the result should be better than the one you show.

yulan0215, 2022-08-31:
Hi:

I also used Ouster LiDAR and AVT-Camera for calibration, can you tell me more specific how to modify the format of Ouster LiDAR point cloud in to Velodyne format? Thx!

unnikrishnanrnair, 2022-08-31:
> If the camera_info has the distortion parameters, the calibration is computed for the undistorted (_rect) image. Undoubtedly, the result should be better than the one you show.

Yes I had forgot to update this here ... Apologies. I projected the points on the rectified image and it looks good. 
[image]

yulan0215, 2022-09-22:
> 

Hi, I also used the OS1 lidar (64 channels) and I would like to change the format of Ouster LiDAR into Velodyne format, do you have any ideas about how to modify the ROS driver of OS1? I used the ROS driver here: https://github.com/ouster-lidar/ouster_example

I will appreciate if you can give me some ideas thx!

yulan0215, 2022-09-22:
> Hi, I am fairly new to ROS but I dont know how to proceed from here or what my mistakes are.
> 
> Setup: The lidar and the camera stand 1.5m from the board. Between board and door there is another 0.5m. [image] [image] [image]
> 
> camera: [ids gv 5250](https://de.ids-imaging.com/store/gv-5250se.html) lidar: [ouster os1](https://ouster.com/products/scanning-lidar/os1-sensor/)
> 
> The camera is monocular. Camera and lidar collect data at 10Hz. I wrote the camera publisher partly myself. Looking like this:
> 
> ` ros::NodeHandle nh; image_transport::ImageTransport it(nh); image_transport::Publisher pub = it.advertise("camera/image", 1);
> 
> ```
> ros::Publisher pub_camera_info = nh.advertise<sensor_msgs::CameraInfo>("camera/camera_info", 1);
> sensor_msgs::CameraInfo camera_info;
> 
> 
> while (m_running)
> {
>     try
>     {
>         // Get buffer from device's datastream
>         const auto buffer = m_dataStream->WaitForFinishedBuffer(5000);
> 
>         QImage qImage(m_imageWidth, m_imageHeight, QImage::Format_RGB32);
> 
>         // Create IDS peak IPL image for debayering and convert it to RGBa8 format
>         const auto image = peak::BufferTo<peak::ipl::Image>(buffer).ConvertTo(
>             peak::ipl::PixelFormatName::BGRa8, qImage.bits(), static_cast<size_t>(qImage.byteCount()));
> 
>         
>         cv::Mat cvimage(qImage.height(),qImage.width(),CV_8UC4, (uchar*)qImage.bits(),qImage.bytesPerLine()); //cv image auf qimage erstellen
>         sensor_msgs::ImagePtr msg = cv_bridge::CvImage(std_msgs::Header(), "bgra8", cvimage).toImageMsg(); //sensor image msg erstellen
> 
>         (*msg).header.frame_id = "camera"; //std::to_string(m_frameCounter);
> 
>         pub.publish(msg); //msg publishen
> 
>         
>         camera_info.header.stamp = ros::Time::now();
>         camera_info.header.frame_id = "camera";
> 
> 
>         camera_info.width = m_nodemapRemoteDevice->FindNode<peak::core::nodes::IntegerNode>("Width")->Value();
>         camera_info.height = m_nodemapRemoteDevice->FindNode<peak::core::nodes::IntegerNode>("Height")->Value();
> 
>         camera_info.distortion_model = "plumb_bob";
> 
> 
>         pub_camera_info.publish(camera_info);
> 
> 
> 
>         // Queue buffer so that it can be used again
>         m_dataStream->QueueBuffer(buffer);
> 
>         // Emit signal that the image is ready to be displayed
>         emit imageReceived(qImage);
> 
>         m_frameCounter++;
>     }
> ```
> 
> `
> 
> So i have an topic camera/image and camera/camera_info which should be the required topics. I read a lot about the frame_id of the headers but i still dont fully understand it. Especially why i need them in this context, since camera_info and the camera/image descripe the same images and thus dont need different coordination system. Anyways, here would be **my first question** if I defined the frame_id correct or if I made a mistake here allready. The header of my lidar data has no frame_id set. Do i need that in this case? Since there is no argument in the launch file i thought i dont need it.
> 
> The point cloud i am publishing in my lidar consist of the following point which is similar to the Velodyne Point:
> 
> struct EIGEN_ALIGN16 Point_xyzir{ PCL_ADD_POINT4D; float intensity; uint16_t ring; float range; EIGEN_MAKE_ALIGNED_OPERATOR_NEW };
> 
> I follow the manual step by step and the launch files work. It says in the manual "For monocular devices, the only relevant data is the image, where the pattern should be fully visible. For range data, the checklist depends on the modality.". So my **second Question** would be if I understand correctly, that i dont have to check my camera image in rviz since i use a monocular device. I tried to check the image in rviz but i get error messages because of my "camera" frame_id that i dont publish anywhere. I get the error: "CameraInfo/P resulted in an invalid position calculation (nans or infs)". I tried to resolve it but it didnt work and i was hoping that this was not the problem bcs of the sentence in the manual mentioned above. [image]
[... 1 lines cut ...]
> Next i try to calibrate the lidar with rviz. The two topics "lidar_pattern_0/zyx_filtered" and "lidar_pattern_0/range_filtered_cloud" that I receive look like this on their own: [image] [image]
> 
> There would be my **third question**: Should they overlap like this? In the manual the datapoints seem to not be overlapping. My topics seem to have identical points. Am i doing something wrong here?
> 
> I manage to be able to cut out the calibration pattern out pretty good with the following settings: [image]
> 
> With this setting the terminal with the lidar launch node outputs following warnings: [ WARN] [1651764290.766055097]: [LiDAR] Not enough centers: 2 [ WARN] [1651764290.867160361]: [LiDAR] Not enough centers: 3 [ WARN] [1651764290.967275613]: [LiDAR] Unable to find a candidate set that matches target's geometry [ WARN] [1651764291.066757160]: [LiDAR] Not enough centers: 0 [ WARN] [1651764291.166676958]: [LiDAR] Not enough centers: 0 [ WARN] [1651764291.265684512]: [LiDAR] Not enough centers: 0 [ WARN] [1651764291.366198506]: [LiDAR] Not enough centers: 0 [ WARN] [1651764291.468210932]: [LiDAR] Unable to find a candidate set that matches target's geometry [ WARN] [1651764291.566586082]: [LiDAR] Not enough centers: 0 [ WARN] [1651764291.666599987]: [LiDAR] Not enough centers: 0 [ WARN] [1651764291.771358378]: [LiDAR] Not enough centers: 1
> 
[... cut ...]
> (sorry for formatting) As i said i am new to ROS and to C++ and cmake aswell. Could this be the problem maybe?
> 
> If I can provide any other information just let me know. I would appreciate any help, thank you.

Hi, I also used the OS1 lidar (64 channels) and I would like to change the format of Ouster LiDAR into Velodyne format, do you have any ideas about how to modify the ROS driver of OS1? I used the ROS driver here: https://github.com/ouster-lidar/ouster_example

I will appreciate if you can give me some ideas thx!

cguindel, 2022-11-21:
Closing the issue as it seems solved.

@yulan0215 Sorry, we don't know how to help you with that; I'm afraid your question is out of the scope of our package.

closed 2022-11-21 by cguindel

yulan0215, 2024-01-11:
> Closing the issue as it seems solved.
> 
> @yulan0215 Sorry, we don't know how to help you with that; I'm afraid your question is out of the scope of our package.

Hello:

When I followed the warmup and do the calibration, the calibration tool did not report the error but stopped at: 
`Please, adjust the filters for each sensor before the "
                "calibration starts."`

I did not know why it show us this problem. Can you give us some ideas for that?