beltransen/velo2cam_calibration #46: verification of the calibration

talhasu, 2022-06-12:
Hi,
Thank you for your wonderful package. I have completed the process and was able to get the following output (see attached screenshot). I am wondering how can I know that the calibration actually worked. Do your package allow live verification? Is there a way to know if the point cloud from lidar is merging with the image from your package?

[image]

cguindel, 2022-06-14:
Currently, we do not provide tools to project the LiDAR point cloud onto the image; instead, we recommend using native ROS apps: if you run the resulting _calibrated_tf.launch_ and your sensors are correctly synchronized, you can use a Camera display in RVIZ to check the result.

talhasu, 2022-06-14:
"you can use a Camera display in RVIZ to check the result." By result, do you mean that we should be able to see the projected cloud on the image? Can you please explain with a screenshot or something? 
Do you mean that I need to keep the calibrated_tf.launch file running and check the camera display?

cguindel, 2022-06-21:
Yes, you need to run the calibrated_tf.launch to broadcast the tf between the two sensors, and then RVIZ will listen to that tf and project the cloud onto the image, as long as the LiDAR cloud is being shown. You can check this example:

[image]

SueeH, 2022-09-07:
> Hi, Thank you for your wonderful package. I have completed the process and was able to get the following output (see attached screenshot). I am wondering how can I know that the calibration actually [...]

Where can I find the test datase( or rosbag)?

cguindel, 2022-11-21:
@SueeH Sorry, we do not provide test data at the moment. 

Closing the issue as it seems solved.

closed 2022-11-21 by cguindel