beltransen/velo2cam_calibration #47: I don't achieve precission claimed in paper with M>3 and M<=5

Nasrudin78, 2022-06-15:
Hello,
I'm trying to calibrate a pinhole camera and HDL32 lidar (both rendered in Gazebo) . I'm using different panels orientations and positions of the included pattern. The result in translation -after registration- is quite good, but in rotation, there are values that are greater than 1 degree -compared with the HDL32 ground truth rendered in Gazebo -.
I've tested with 3 up to 5 different pattern poses,tilts and distances (avoiding co-planarity) but without any success in lowering rotation differences below 1 degree.
Is it possible to minimize even more the difference < 1 degree? 
Thanks for your contribution and help.

cguindel, 2022-06-21:
It is difficult for us to guess the possible causes of the discrepancy you have observed in your tests, as many factors can be involved. One of them is the location of the calibration pattern: if it is too far away from the sensors, the performance may be degraded. To discard other sources of error, maybe you can try to replicate the Single-Sensor Experiments from our T-ITS paper (the relative target-sensor positions that we used are reported in Table III) and make sure that the errors in the location of the reference points are within the range of those shown in Figures 4 and 5.

Nasrudin78, 2022-07-01:
Hi,
Finally I discovered the error source. I swapped values in  Yaw and Roll rotation components during the basis change process.
With only 1 panel the error is below 0.2 degrees in the rotation vector.
Thanks for your help!

closed 2022-07-01 by Nasrudin78