beltransen/velo2cam_calibration #49: Calibration results are of very low quality

alasdairhunter, 2022-06-21:
I am trying to calibrate a rgb monocular camera and a velodyne vlp32 camera. I have followed the instructions on the github, and everything works well as far as I can tell, but the output is basically just nonsense. 
The camera is located directly beneath the lidar with a vertical displacement of about 10 cm, but the calibration results vary a lot whenever the target is moved, and none of the results are good. I am talking about displacements of well over 20 cm.
I dont have any idea why this should be. Has anyone else encountered very poor results, then realized it was do to a specific parameter?
Range filtered cloud:
[image]
XYZ filtered cloud:
[image]
Camera:
[image]

Mkarasneh, 2022-06-21:
Hello @alasdairhunter, I am also trying to calibrate a monocular camera with the velodyne VLP 32. However, I didn't figure out how to visualize the output as in the last picture in your question, can you share with me how you obtained that output?

cguindel, 2022-06-21:
@alasdairhunter We have never observed such a large error in our tests. I guess the problem is related to your specific configuration, but I can give you some advice to troubleshoot the issue:
- Have you tried using more than one pattern pose (M>1)?
- Are the intrinsic parameters of your fish-eye camera properly calibrated?
- As it seems that you have already activated the DEBUG mode, could you look at the _cumulative_cloud_ topics in RVIZ? They contain the center locations of the pattern circles in both modalities, so they should look like this:

[image]

@Mkarasneh You can obtain that visualization by setting the DEBUG flag to 1 [here](https://github.com/beltransen/velo2cam_calibration/blob/bfba57837e1efd979f8514512756b450f921e2aa/include/velo2cam_utils.h#L30).

alasdairhunter, 2022-06-23:
@Mkarasneh set DEBUG to 1 in include/velo2cam_utils.h

lbouz102, 2022-08-09:
@alasdairhunter Did you change tune any parameters besides pass_through_radius_[min/max] and pass_through_[x,y,z]_[min/max]? My zyx and range filtered clouds look similar to yours but I'm having the issue described in #51

cguindel, 2022-11-21:
Closing the issue due to inactivity. Feel free to reopen it if you need.