beltransen/velo2cam_calibration #50: Calibration Accuracy

Mkarasneh, 2022-07-03:
Hello,

I have tried using the package to calibrate a velodyne VLP-32c Lidar with an Allied vision camera. The output for the code are as follows:
 
**1. mono_camera_pattern** 
```[ INFO] [1656874411.241321437]: [Mono] 1148/1148 frames: 4592 pts in cloud
4 clusters found from 4592 points in cloud
4 centers found. Iterating over 1 possible sets of candidates
[ INFO] [1656874412.408262713]: [Mono] 1149/1149 frames: 4596 pts in cloud
4 clusters found from 4596 points in cloud
4 centers found. Iterating over 1 possible sets of candidates
[ INFO] [1656874413.579516083]: [Mono] 1150/1150 frames: 4600 pts in cloud
4 clusters found from 4600 points in cloud
4 centers found. Iterating over 1 possible sets of candidates
[ INFO] [1656874414.766473655]: [Mono] 1151/1151 frames: 4604 pts in cloud
4 clusters found from 4604 points in cloud
4 centers found. Iterating over 1 possible sets of candidates
[ INFO] [1656874415.949315396]: [Mono] 1152/1152 frames: 4608 pts in cloud
4 clusters found from 4608 points in cloud
4 centers found. Iterating over 1 possible sets of candidates
[ INFO] [1656874417.120503392]: [Mono] 1153/1153 frames: 4612 pts in cloud
4 clusters found from 4612 points in cloud
4 centers found. Iterating over 1 possible sets of candidates
[ INFO] [1656874418.301987090]: [Mono] 1154/1154 frames: 4616 pts in cloud
4 clusters found from 4616 points in cloud
4 centers found. Iterating over 1 possible sets of candidates
[ INFO] [1656874419.493276712]: [Mono] 1155/1155 frames: 4620 pts in cloud
4 clusters found from 4620 points in cloud
4 centers found. Iterating over 1 possible sets of candidates
```

**2. Lidar_pattern**
```[ INFO] [1656874912.543768664]: [LiDAR] Searching for points in cloud of size 106
[ INFO] [1656874912.546013116]: [LiDAR] Remaining points in cloud 84
[ INFO] [1656874912.547459386]: [LiDAR] Remaining points in cloud 63
[ INFO] [1656874912.550581930]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
[ INFO] [1656874912.553221275]: [LiDAR] Remaining points in cloud 50
[ INFO] [1656874912.556925593]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
[ INFO] [1656874912.559266546]: [LiDAR] Remaining points in cloud 36
[ INFO] [1656874912.563045375]: [LiDAR] Remaining points in cloud 30
[ INFO] [1656874912.565608706]: [LiDAR] Remaining points in cloud 24
[ INFO] [1656874912.567419382]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
[ INFO] [1656874912.568163642]: [LiDAR] Remaining points in cloud 17
[ INFO] [1656874912.569939733]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
[ INFO] [1656874912.570829697]: [LiDAR] Remaining points in cloud 9
[ INFO] [1656874912.575357324]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
8 centers found. Iterating over 70 possible sets of candidates
[ WARN] [1656874912.579125302]: [LiDAR] Unable to find a candidate set that matches target's geometry
[ INFO] [1656874912.616781595]: [LiDAR] Processing cloud...
Rot. vector: -0.0249148
  -0.99969
        -0 / Angle: 1.57718
[ INFO] [1656874912.622147709]: [LiDAR] Searching for points in cloud of size 106
[ INFO] [1656874912.624790129]: [LiDAR] Remaining points in cloud 84
[ INFO] [1656874912.626621083]: [LiDAR] Remaining points in cloud 63
[ INFO] [1656874912.630380615]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
[ INFO] [1656874912.633573661]: [LiDAR] Remaining points in cloud 51
[ INFO] [1656874912.637495809]: [LiDAR] Remaining points in cloud 45
[ INFO] [1656874912.640966578]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
[ INFO] [1656874912.643865611]: [LiDAR] Remaining points in cloud 31
[ INFO] [1656874912.646313826]: [LiDAR] Remaining points in cloud 25
[ INFO] [1656874912.648415229]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
[ INFO] [1656874912.649344497]: [LiDAR] Remaining points in cloud 16
[ INFO] [1656874912.650064632]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
[ INFO] [1656874912.650328228]: [LiDAR] Remaining points in cloud 9
[ INFO] [1656874912.653778507]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
8 centers found. Iterating over 70 possible sets of candidates
[ INFO] [1656874912.657597802]: [LiDAR] 1334/1887 frames: 5336 pts in cloud
4 clusters found from 5336 points in cloud
[ INFO] [1656874914.585150382]: [LiDAR] Pattern centers published
[ INFO] [1656874914.585316674]: [LiDAR] Processing cloud...
Rot. vector: -0.0215657
 -0.999767
        -0 / Angle: 1.57745
```

**3. Registration**
```
30/30 iterations
Target iterations reached. Do you need another target location? [y/N]
N
Calibration finished succesfully.
Extrinsic parameters:
x = 0.0860	y = -0.3071	z = -0.0359
roll = 0.0070	pitch = -0.0329	yaw = -0.0414
```

Here is the visualization result in RVIZ:
[image]
[image]
[image]
[image]

**It is worth noting that I have tried changing the camera position and orientation, yet every time the calibration result is the same, the lower left circle gets aligned with the upper right circle. I am not sure what the problem can be. Also, the positioning of the camera and lidar TF in RVIZ makes sense compared to my actual setup, but the overlay doesn't**

Thank you!

lbouz102, 2022-08-09:
Did you end up solving this? What does your /lidar_patter_0/pattern_circles cloud look like? Is it sparse similar to #51?

Mkarasneh, 2022-08-17:
> @Mkarasneh Did you end up solving this? What does your /lidar_patter_0/pattern_circles cloud look like? Is it sparse similar to #51?

@lbouz102 No, the pattern circle is filtered nicely, not like #51. I have attempted a different time with the target closer, resulting in more dense lidar data on the target. There is still an error in the rotation despite the fact that this partially reduced the shift. I had to manually adjust the yaw, pitch, and roll angles for them to be merged. I'm unsure if it has anything to do with the fabrication precision of the wooden target, but that shouldn't be a problem given that the code specifies a tolerance. Anyway, I will attempt again this month and update this thread with my results.

SueeH, 2022-09-07:
> Hello,
> 
> I have tried using the package to calibrate a velodyne VLP-32c Lidar with an Allied vision camera. The output for the code are as follows:
> 
> **1. mono_camera_pattern**
> 
> ```
> 4 clusters found from 4592 points in cloud
> 4 centers found. Iterating over 1 possible sets of candidates
> [ INFO] [1656874412.408262713]: [Mono] 1149/1149 frames: 4596 pts in cloud
> 4 clusters found from 4596 points in cloud
> 4 centers found. Iterating over 1 possible sets of candidates
> [ INFO] [1656874413.579516083]: [Mono] 1150/1150 frames: 4600 pts in cloud
> 4 clusters found from 4600 points in cloud
> 4 centers found. Iterating over 1 possible sets of candidates
> [ INFO] [1656874414.766473655]: [Mono] 1151/1151 frames: 4604 pts in cloud
> 4 clusters found from 4604 points in cloud
> 4 centers found. Iterating over 1 possible sets of candidates
> [ INFO] [1656874415.949315396]: [Mono] 1152/1152 frames: 4608 pts in cloud
> 4 clusters found from 4608 points in cloud
> 4 centers found. Iterating over 1 possible sets of candidates
> [ INFO] [1656874417.120503392]: [Mono] 1153/1153 frames: 4612 pts in cloud
> 4 clusters found from 4612 points in cloud
> 4 centers found. Iterating over 1 possible sets of candidates
> [ INFO] [1656874418.301987090]: [Mono] 1154/1154 frames: 4616 pts in cloud
> 4 clusters found from 4616 points in cloud
> 4 centers found. Iterating over 1 possible sets of candidates
> [ INFO] [1656874419.493276712]: [Mono] 1155/1155 frames: 4620 pts in cloud
> 4 clusters found from 4620 points in cloud
> 4 centers found. Iterating over 1 possible sets of candidates
> ```
> 
> **2. Lidar_pattern**
> 
> ```
> [ INFO] [1656874912.546013116]: [LiDAR] Remaining points in cloud 84
> [ INFO] [1656874912.547459386]: [LiDAR] Remaining points in cloud 63
> [ INFO] [1656874912.550581930]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> [ INFO] [1656874912.553221275]: [LiDAR] Remaining points in cloud 50
> [ INFO] [1656874912.556925593]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> [ INFO] [1656874912.559266546]: [LiDAR] Remaining points in cloud 36
> [ INFO] [1656874912.563045375]: [LiDAR] Remaining points in cloud 30
> [ INFO] [1656874912.565608706]: [LiDAR] Remaining points in cloud 24
> [ INFO] [1656874912.567419382]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> [ INFO] [1656874912.568163642]: [LiDAR] Remaining points in cloud 17
> [ INFO] [1656874912.569939733]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> [ INFO] [1656874912.570829697]: [LiDAR] Remaining points in cloud 9
> [ INFO] [1656874912.575357324]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> 8 centers found. Iterating over 70 possible sets of candidates
> [ WARN] [1656874912.579125302]: [LiDAR] Unable to find a candidate set that matches target's geometry
> [ INFO] [1656874912.616781595]: [LiDAR] Processing cloud...
> Rot. vector: -0.0249148
>   -0.99969
>         -0 / Angle: 1.57718
> [ INFO] [1656874912.622147709]: [LiDAR] Searching for points in cloud of size 106
> [ INFO] [1656874912.624790129]: [LiDAR] Remaining points in cloud 84
> [ INFO] [1656874912.626621083]: [LiDAR] Remaining points in cloud 63
> [ INFO] [1656874912.630380615]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> [ INFO] [1656874912.633573661]: [LiDAR] Remaining points in cloud 51
> [ INFO] [1656874912.637495809]: [LiDAR] Remaining points in cloud 45
> [ INFO] [1656874912.640966578]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> [ INFO] [1656874912.643865611]: [LiDAR] Remaining points in cloud 31
> [ INFO] [1656874912.646313826]: [LiDAR] Remaining points in cloud 25
> [ INFO] [1656874912.648415229]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> [ INFO] [1656874912.649344497]: [LiDAR] Remaining points in cloud 16
> [ INFO] [1656874912.650064632]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> [ INFO] [1656874912.650328228]: [LiDAR] Remaining points in cloud 9
> [ INFO] [1656874912.653778507]: [LiDAR] Optimized circle segmentation failed, trying unoptimized version
> 8 centers found. Iterating over 70 possible sets of candidates
> [ INFO] [1656874912.657597802]: [LiDAR] 1334/1887 frames: 5336 pts in cloud
> 4 clusters found from 5336 points in cloud
> [ INFO] [1656874914.585150382]: [LiDAR] Pattern centers published
> [ INFO] [1656874914.585316674]: [LiDAR] Processing cloud...
> Rot. vector: -0.0215657
>  -0.999767
>         -0 / Angle: 1.57745
> ```
> 
> **3. Registration**
> 
> ```
> 30/30 iterations
> Target iterations reached. Do you need another target location? [y/N]
> N
> Calibration finished succesfully.
> Extrinsic parameters:
> x = 0.0860	y = -0.3071	z = -0.0359
> roll = 0.0070	pitch = -0.0329	yaw = -0.0414
> ```
> 
> Here is the visualization result in RVIZ: [image] [image] [image] [image]
> 
> **It is worth noting that I have tried changing the camera position and orientation, yet every time the calibration result is the same, the lower left circle gets aligned with the upper right circle. I am not sure what the problem can be. Also, the positioning of the camera and lidar TF in RVIZ makes sense compared to my actual setup, but the overlay doesn't**
> 
> Thank you!
I have compiled the program, where is the test datasets?

Willforcv, 2022-09-21:
how did you visualize colored point cloud ?

morgan9900103, 2022-09-25:
Did you check if the camera intrinsic is correct? From my past experience there might be a chance that the issue is caused by that

Mkarasneh, 2022-10-10:
> how did you visualize colored point cloud ?

You can do that in rviz. First, launch the calibrated_tf.launch. Second add the pointcloud2 and camera, set the camera and lidar topic accordingly. Then change the fixed frame to the rotated camera.

Mkarasneh, 2022-10-10:
> Did you check if the camera intrinsic is correct? From my past experience there might be a chance that the issue is caused by that

@morgan9900103. I used the [ros camera calibrator package](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) to calibrate the camera. I am not sure how to check if the matrices are correct. I think that makes sense, if you any suggestions regarding calibrating the camera and checking the results please let me know.

Mkarasneh, 2022-11-17:
> 

Thank you for your insight. It turned out that the intrinsic matrix was not accurate, fixing it made the results of the fusion match

closed 2022-11-17 by Mkarasneh