beltransen/velo2cam_calibration #51: pattern_circles point cloud is sparse

lbouz102, 2022-07-20:
Raw point cloud:
[image]

ZYX filtered cloud:
[image]

Pattern circles:
[image]

I'm trying to understand what's wrong with my pattern circles. The log 7 candidate circles are found but it doesn't find a set that matches geometry. Visually my raw and filtered clouds appear to be fine but the pattern cloud not so much. Any ideas to what could be going wrong here? I've changed a few distances to match my board (aruco markers relative to hole centers and hole radii).

cguindel, 2022-08-22:
I don't know what is happening here, but I would like to figure it out. Could you send us a bag file with a snippet of your data so we can reproduce the issue? If you prefer, you can send it by email; my address is in the [paper](https://arxiv.org/abs/2101.04431).

lbouz102, 2022-08-30:
It ended up being an issue with how I was sending the point cloud. I misunderstood the requirement that the ring Ids were necessary in order to search for the edges. Since all my points were labeled ring 0 it was detecting "random" points as edges. My pattern circle cloud now looks much better:
[image]

I am still having an issue where > 4 circles are being detected and it doesn't fit the target geometry ~95% of the time. The screenshot above is good example of one where the hole centers are often not detected. Do you have any suggestions for parameters I can tune or is this just a product of the 32-channel VLP not having enough points?

cguindel, 2022-08-31:
Thanks for your reply. Interestingly, that is not the behavior we observed when all the points were assigned ring 0. In that case, every point is treated as if they belong to a single ring, and the search for "jumps" is done along that very long ring, which could cause some noise, but nothing that problematic. Is it possible that the ring values were just assigned randomly before?

Regarding your current result, the VLP-32 should definitely be dense enough for the method to work. However, your pattern_circles cloud does not look as expected. It should be something similar to this:

[image]

That is, after the edge extraction procedure, there should be a single point representing the edges of the pattern, as we do not perform accumulation at that stage. However, in your case, it seems like you have small clusters instead. So I think the source of the problem could still be in the data, but I can't say if it is still related to the ring field or comes from elsewhere.

lbouz102, 2022-08-31:
I didn't define the rings field in my sensor_msg so its possible that it was randomly assigned to whatever values were in memory? I'll look at this again to clarify for you once I've solved the other issue.

Yes, you're right that I have small clusters. I aggregated the Velodyne data before sending it because I thought it would help with the circle detection. After your response I realize this isn't the correct approach so I've removed that and I'm still having the same issues. Here is my pattern circle cloud now:
[image]
The only way I'm able to get some centers correct is if I increase my target_radius_tolerance to 0.02. I've remeasured my target to ensure the dimensions are correct.

cguindel, 2022-11-21:
Hi @lbouz102, could you solve the issue? Did you obtain good results after tuning the target_radius_tolerance parameter?

lbouz102, 2022-11-22:
This issue was resolved by reducing the RPM of the velodyne (1200 to 300). The edge points on the circle didn't form a clean circle as shown in the picture above. Feel free to close this issue as the initial post was a misunderstanding that an ordered point cloud is needed. Thanks!

cguindel, 2022-11-23:
Ok, thank you for your reply; it is good to be aware of that caveat for future reference. 

Closing the issue as suggested.

closed 2022-11-23 by cguindel