beltransen/velo2cam_calibration #52: I have a question about the Aruco placement

nyxrobotics, 2022-08-23:
I am working according to the howto documentation, but have a question.
I follow the sample diagram to create a board for calibration.
However, the Aruco marker is larger than the sample photo, and the dimensions of the margins do not seem to match.
Am I wrong about the margins?

--

[image]

beltransen, 2022-08-23:
HI @nyxrobotics. You are right. In the photo, the margins are a bit larger than in the dimensional drawing. This happens because we were testing boards of different materials and realized that the use of a white margin improved the precision in the detection of the Aruco markers when using a non-white calibration target.

However, as stated in the footnote, the dimensions of the board and the placement of the holes and markers can be changed as long as you adapt the corresponding parameters in the launch file accordingly. On the other hand, if you are using a white board we recommend removing these white margins to have bigger markers so the monocular detection at larger distances is more accurate.

nyxrobotics, 2022-08-31:
Thanks for the reply, it worked.
This package did not recognize the circle well without a wall behind the calibration board.
Placing a flat board about 300mm behind the calibration board worked well to recognize it.

closed 2022-08-31 by nyxrobotics